alexdelprete/ha-abb-powerone-pvi-sunspec #22: Dynamic Feed-In Control Sensor

gotttlieb, 2022-04-12:
Hi there. 
I'am using an inverter UNO-DM-4.0-TL-PLUS  with Dynamic Feed-In Control via REACT-MTR-1PH but it seems that the app is not loggin the Grid useage as in the Inverter APP/Web portal.
This will be pretty handy to check and keep my grid usage and total energy consumtion (including the solar).
[image]

I don't know if its feasible.
Thanks for advance for your time.

alexdelprete, 2022-05-26:
> it seems that the app is not loggin the Grid useage as in the Inverter APP/Web portal.

the component was developed based on the register map of the inverter. so it connects to the inverter via modbus and queries those registers. in the map I analyzed, I didn't see registers for the meters. I don't know if that would require a separate modbus connection to the meter or if the inverter you use provides an extended register map with the meter data included.

Since I don't have that meter and that inverter, nor the relevant documentation, I cannot try to implement it because I couldn't test it, but you can always try to do it and submit a PR if you like.

Sorry about that, I wish I could help you more than this.

closed 2022-10-05 by alexdelprete

gotttlieb, 2022-10-06:
I really appreciate you time to respond to my question.
For all the people that are in the same situation as me, at least in Spain, we have an HA integration 
e-data that works with datadis. This integration is not in real time, but it shows the real value of kw and hour distribution. ( sometimes delays can be up to 5-6 days in my case) .
But it works for me. 
Thanks for the awesome integration and hard work.

alexdelprete, 2022-10-06:
Does your inverte use a VSNx00 card for the sunspec modbus tcp interface?

gotttlieb, 2022-10-11:
I check in the provider webpage, and i think so. But i'am connected using Ethernet cable. ( but i have the wifi option an i already check it at first installation)