alexdelprete/ha-abb-powerone-pvi-sunspec #31: Unable to import component: No module named 'pymodbus.client.sync'  after upgrade to 2022.10.3 and/or OS 9.2

Claudio1L, 2022-10-14:
recently I started to see error msgs after the reboot. Usually after one or two try the error disapper and working fine. Today I'm not able to make it works.
Looks like a module is missing. Could be one of recent updates?

2022-10-14 08:22:57.839 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration abb_powerone_pvi_sunspec: Unable to import component: No module named 'pymodbus.client.sync'

Claudio1L, 2022-10-14:
full log here:

2022-10-14 10:02:44.745 ERROR (MainThread) [homeassistant.loader] Unexpected exception importing component custom_components.abb_powerone_pvi_sunspec
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/loader.py", line 703, in get_component
cache[self.domain] = importlib.import_module(self.pkg_path)
File "/usr/local/lib/python3.10/importlib/__init__.py", line 126, in import_module
return _bootstrap._gcd_import(name[level:], package, level)
File "<frozen importlib._bootstrap>", line 1050, in _gcd_import
File "<frozen importlib._bootstrap>", line 1027, in _find_and_load
File "<frozen importlib._bootstrap>", line 1006, in _find_and_load_unlocked
File "<frozen importlib._bootstrap>", line 688, in _load_unlocked
File "<frozen importlib._bootstrap_external>", line 883, in exec_module
File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
File "/config/custom_components/abb_powerone_pvi_sunspec/__init__.py", line 9, in <module>
from pymodbus.client.sync import ModbusTcpClient
File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 166, in <module>
class ModbusTcpClient(BaseModbusClient):
File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 170, in ModbusTcpClient
def __init__(self, host='127.0.0.1', port=Defaults.Port,
AttributeError: type object 'Defaults' has no attribute 'Port'
2022-10-14 10:02:44.749 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration abb_powerone_pvi_sunspec: Unable to import component: Exception importing custom_components.abb_powerone_pvi_sunspec

Claudio1L, 2022-10-14:
looks like python disappeared?
[image]

Claudio1L, 2022-10-14:
reverted back to a snapshot with OS 9.0 and it worked.

It is something related to OS 9.2

Claudio1L, 2022-10-14:
Same issue after reboot with 9.0 Also.

seems related to 3.0.0 pymodbus recently updated
https://github.com/home-assistant/operating-system/issues/2188

alexdelprete, 2022-10-14:
> seems related to 3.0.0 pymodbus recently updated

The component uses a pinned version of pymodbus, so the upgrade of pymodbus should not have effect.

```json
{
  "domain": "abb_powerone_pvi_sunspec",
  "name": "ABB Power-One PVI SunSpec",
  "documentation": "https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec",
  "issue_tracker": "https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/issues",
  "requirements": ["pymodbus==1.5.2"],
  "dependencies": [],
  "codeowners": ["@alexdelprete"],
  "iot_class": "local_polling",
  "config_flow": true,
  "version": "1.0"
}
```

Claudio1L, 2022-10-14:
Any advice on how to proceed ?

Have same issue in two integrations using same component.
The second one is:

https://github.com/wills106/homeassistant-solax-modbus/issues/154 <https://github.com/wills106/homeassistant-solax-modbus/issues/154>

Thanks
Claudio

alexdelprete, 2022-10-14:
Maybe I understood the problem:the solax integration doesn't use a pinned version of pymodbus, so this could probably install pymodbus 3.0, that is not compatible with my integration, and I think that is not compatible also with solax integration because 3.0 is a breaking change release.

Test this: uninstall both integrations and restart. then reinstall mine (ABB) and restart, check if it works. If it does, then the problem is the conflict on the pymodbus version requirement of the two integrations.

Claudio1L, 2022-10-15:
Don't understand why but it was a conflict with the Bluetooth usb key I recently added. Removing the key and the bluetooth integration is working consistently fine. At the moment I can live without it. My apologise for the confusion

closed 2022-10-15 by Claudio1L

Claudio1L, 2022-10-20:
after update to 2.1.0 issue appeared again on bot integration. I'm lost, I had to rollback to the previous snapshot. Before rolling back, I tried to remove everything form integration page and redownloading from hacs, ABB integration (2.1.0) was working fine, Solax integration refused to install for missing pymodbus.client.sync. I'll reopen the bug on Solax beacause it seems it have issue with pymodbus 3.0.0.

alexdelprete, 2022-10-20:
You have to tell other devs to update their components to latest pymodbus 3.0. That's what I did.

Claudio1L, 2022-10-21:
Confirmed. They just released a beta to address this issue and seems working fine. Thank you for the support