alexdelprete/ha-abb-powerone-pvi-sunspec #32: Model unknown Manuf.: ABB - Model: -3P46-

Claudio1L, 2022-10-14:
After long time running fine, today started loggin this message of model unknown

2022-10-14 13:48:43.663 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] (opt_comm_model) Model unknown, report to @alexdelprete on the forum the following data: Manuf.: ABB - Model: -3P46- - Options: 0x03ED/0xFF69/ - OptModel: 0x03ED/0xFF69/ - OptModelInt: 3

alexdelprete, 2022-10-14:
Claudio, it's not a real error, did you read the message? :)

I put the error on purpose to ask users to provide the real model of the inverter associated with Option Model (in your case it's 3) so I can add it to the list of inverters known to the integration.

So, can you tell me the specific model of your inverter? In next update you'll find the correct model and no error in the log.

Ciao.

Claudio1L, 2022-10-14:
Thanks

The mode il UNO-DM-4.6-TL-PLUS

Thanks

Claudio

alexdelprete, 2022-10-14:
Added your model and released new version. If you upgrade the component, you should see the model in the sensors list, instead of 0x03ED/0xFF69.

Let me know if it works.

Claudio1L, 2022-10-14:
Thank you. I’m worried to reboot due to the other issue I opened today, the one relate do pymodbus

alexdelprete, 2022-10-14:
> Thank you. I’m worried to reboot due to the other issue I opened today, the one relate do pymodbus

Sorry, I didn't check that issue. I'll answer there. Closing this issue.

closed 2022-10-14 by alexdelprete

Claudio1L, 2022-10-14:
Just tested doing a snapshot before reboot and nothing. Had to restore the snapshot because of the issue #31 preventing from load the integration