alexdelprete/ha-abb-powerone-pvi-sunspec #44: Model unknown Manuf.: ABB - Model: -3P25-

usrpend, 2022-12-04:
Hello Claudio,

Could you please add the model FIMER UNO-DM-5.0-TL-PLUS to the database, which is identified with the code -3P25-?
Also, I'm facing an issue due to an error following the installation of the latest version (2.2.1) due to pymodbus not working. Could you please help me in solving this, please?
Thanks in advance for your assistance
Domenico

issuluca, 2022-12-07:
Hi,
I also have the same problem, until version 2.0xx everything worked correctly, now it doesn't work anymore, and I can't even delete the device.
Thanks
Luca

alexdelprete, 2023-01-09:
> Could you please add the model FIMER UNO-DM-5.0-TL-PLUS to the database, which is identified with the code -3P25-?

Ciao Domenico. As explained on the [forum](https://community.home-assistant.io/t/custom-component-abb-power-one-fimer-pv-inverters-sunspec-modbus-tcp/316363/171?u=alexdelprete), I need the info from the log, you should find this kind of "error" with the data needed to update the list:

```
(opt_comm_model) Model unknown, report to @alexdelprete on the forum the following data: Manuf.: ABB - Model: -3P31- - Options: 0x10ED/0xFF69/ - OptModel: 0x10ED/0xFF69/ - OptModelInt: 16
```

> Also, I'm facing an issue due to an error following the installation of the latest version (2.2.1) due to pymodbus not working. Could you please help me in solving this, please?

You need to provide me more info. "not working" doesn't help me understand what your problem is. What error do you have when installing? When do you have the error, when installing or configuring it? Or after the configuration? Try to be detailed.

Anyway, try latest version I just released: v2.3.0 and let me know. Also, make sure you're not using the modbus native integration in config.yml, other users have had [problems](https://community.home-assistant.io/t/custom-component-abb-power-one-fimer-pv-inverters-sunspec-modbus-tcp/316363/196?u=alexdelprete) because they forgot to disable modbus integration when they switched to this component.

alexdelprete, 2023-01-09:
> I also have the same problem, until version 2.0xx everything worked correctly, now it doesn't work anymore, and I can't even delete the device.

"Doesn't work" will not allow me to help understanding what's your problem. You have to be more specific, provide log errors etc.

Also, your problem might not be the same Domenico has. Open a separate issue if needed.

usrpend, 2023-01-22:
Model: -3P25-
Options: 0x0FED/0xFF69/

alexdelprete, 2023-01-22:
Could you paste here the log lines from HA please? They contain some needed values.

usrpend, 2023-01-22:
2023-01-22 16:13:18.632 ERROR (MainThread) [pymodbus.client.sync] Connection to (192.168.1.102, 502) failed: timed out
2023-01-22 16:13:18.654 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Manufacturer: ABB                             
2023-01-22 16:13:18.655 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Model: -3P25-                          
2023-01-22 16:13:18.657 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Options: 0x0FED/0xFF69/  
2023-01-22 16:13:18.671 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Slave ID: 1
2023-01-22 16:13:18.673 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Base Address: 40000
2023-01-22 16:13:18.674 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Inverter Type: 101
2023-01-22 16:13:18.676 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec.sensor] Model: -3P25-                          
2023-01-22 16:13:18.678 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec.sensor] manufacturer: ABB                             
2023-01-22 16:13:18.679 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec.sensor] SW Version: 2201A           
2023-01-22 16:13:18.681 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec.sensor] Inverter Type: Single Phase
2023-01-22 16:13:18.703 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up abb_powerone_pvi_sunspec platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 281, in _async_setup_platform
    await asyncio.shield(task)
  File "/config/custom_components/abb_powerone_pvi_sunspec/sensor.py", line 20, in async_setup_entry
    hub.read_modbus_data_inverter()
  File "/config/custom_components/abb_powerone_pvi_sunspec/__init__.py", line 243, in read_modbus_data_inverter
    inverter_data = self.read_holding_registers(unit=self._slave_id, address=(self._base_addr + 4), count=64)
  File "/config/custom_components/abb_powerone_pvi_sunspec/__init__.py", line 181, in read_holding_registers
    return self._client.read_holding_registers(address, count, **kwargs)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/common.py", line 114, in read_holding_registers
    return self.execute(request)
  File "/usr/local/lib/python3.10/site-packages/pymodbus/client/sync.py", line 108, in execute
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.1.102:502)]
2023-01-22 16:13:38.734 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Manufacturer: ABB                             
2023-01-22 16:13:38.736 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Model: -3P25-                          
2023-01-22 16:13:38.737 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Options: 0x0FED/0xFF69/  
2023-01-22 16:13:38.750 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Slave ID: 1
2023-01-22 16:13:38.752 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Base Address: 40000
2023-01-22 16:13:38.754 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Inverter Type: 101
2023-01-22 16:13:58.723 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Manufacturer: ABB                             
2023-01-22 16:13:58.724 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Model: -3P25-                          
2023-01-22 16:13:58.725 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Options: 0x0FED/0xFF69/  
2023-01-22 16:13:58.743 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Slave ID: 1
2023-01-22 16:13:58.744 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Base Address: 40000
2023-01-22 16:13:58.745 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Inverter Type: 101
2023-01-22 16:14:21.716 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Reading data failed! Please check Slave ID: 1
2023-01-22 16:14:21.717 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Reading data failed! Please check Reg. Base Address: 40000
2023-01-22 16:14:25.514 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Manufacturer: ABB                             
2023-01-22 16:14:25.515 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Model: -3P25-                          
2023-01-22 16:14:25.516 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Options: 0x0FED/0xFF69/  
2023-01-22 16:14:25.544 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Slave ID: 1
2023-01-22 16:14:25.545 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Base Address: 40000
2023-01-22 16:14:25.546 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Inverter Type: 101
2023-01-22 16:14:25.547 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec.sensor] Model: -3P25-                          
2023-01-22 16:14:25.548 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec.sensor] manufacturer: ABB                             
2023-01-22 16:14:25.549 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec.sensor] SW Version: 2201A           
2023-01-22 16:14:25.549 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec.sensor] Inverter Type: Single Phase
[... cut ...]
2023-01-22 16:14:41.788 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Model: -3P25-                          
2023-01-22 16:14:41.790 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Options: 0x0FED/0xFF69/  
2023-01-22 16:14:44.794 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Slave ID: 1
2023-01-22 16:14:44.795 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Base Address: 40000
2023-01-22 16:14:44.796 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Reading data failed! Please check Slave ID: 1

alexdelprete, 2023-01-22:
Do you have in the logs lines that contain text that says to contact me to report the model etc.? Search for "alexdelprete" in the log.

I see from the log that your component doesn't connect, you didn't solve the problem? Is the Slave ID correct? Is the Base Address correct?

usrpend, 2023-01-22:
Alex, I searched the logs but I'm not able to find "alexdelprete" in it.
I'm currently running v. 2.0.5-beta.2 which is quite old, but I did not update because of the errors I was receiving with newer versions.
The slave ID 1 seems to be correct, communication is working and also QModMaster is able to retrieve values starting address 40000

alexdelprete, 2023-01-22:
That log above reports 3 different READ DATA ERRORS. It means it wasn't able to contact the inverter (or the VSN300). How is it working? It happens often?

usrpend, 2023-01-22:
No, it was today, due to an update of HACS software. After that, all is working correctly.

alexdelprete, 2023-01-22:
Did you try latest version? I'm also about to release a newer version that allows to configure the parameters at runtime. Try it and let me know...old versions are not recommended anymore. :)

usrpend, 2023-01-22:
Thanks Alex, I will try and let you know in case I need your great assistance

alexdelprete, 2023-01-22:
No problem. Let me know if something doesn't work.

alexdelprete, 2023-01-22:
> UNO-DM-5.0-TL-PLUS

Added your model to the known list. It'll be included in next release.

alexdelprete, 2023-01-23:
> 2023-01-23 11:00:19.614 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
> 2023-01-23 11:00:19.616 WARNING (SyncWorker_3) [homeassistant.loader] We found a custom integration abb_powerone_pvi_sunspec which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

These are not an issue, it's a standard msg from HA when you use a custom component that is not an official integration. No worries.

> 2023-01-23 11:00:41.597 ERROR (MainThread) [homeassistant.setup] Setup failed for modbus: Unable to import component: No module named 'pymodbus.client.sync'
> 
> Also, another device which uses modbus does not work, seemingly due to some sort of conflict.

Another user reported this, he was using the component and also modbus: integration, and they conflicted (I don't know why, I used the two together in the past with no issues).

Try disabling the modbus: integration for now, commenting the lines of the other device in the config, and restart HA, so we're sure my component is working. Once you verify it's working, readd the modbus: integration for the other device and restart HA, tell me what happens.

BTW: I didn't release the new version with your model, how can it show up? Let me see the list of the sensors with the model so I can check how it's identified. Thanks.

alexdelprete, 2023-01-23:
@usrpend please open a separate issue about the modbus: conflict.

I'll close this one, ok?

closed 2023-01-23 by alexdelprete