alexdelprete/ha-abb-powerone-pvi-sunspec #46: Config flow - error 500 Internal Server Error

chrisvor, 2023-01-09:
Running HASSOS on Raspberry Pi 4.  Added through HACS, restarted HA, and proceeded to add integration in HA. 
HA on latest version 2023.01.04

Getting error:
Config flow could not be loaded: 500 Internal Server Error Server got itself in trouble

Any idea how to fix?

Log:
Logger: aiohttp.server
Source: config_entries.py:865
First occurred: 12:03:47 AM (8 occurrences)
Last logged: 12:27:53 AM

Error handling request
Traceback (most recent call last):
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_protocol.py", line 435, in _handle_request
    resp = await request_handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_app.py", line 504, in _handle
    resp = await handler(request)
  File "/usr/local/lib/python3.10/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 60, in security_filter_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 100, in forwarded_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 28, in request_context_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 81, in ban_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 236, in auth_middleware
    return await handler(request)
  File "/usr/src/homeassistant/homeassistant/components/http/view.py", line 136, in handle
    result = await result
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 148, in post
    return await super().post(request)
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 73, in wrapper
    result = await method(view, request, data, *args, **kwargs)
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 71, in post
    result = await self._flow_mgr.async_init(
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 229, in async_init
    flow, result = await task
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 247, in _async_init
    flow = await self.async_create_flow(handler, context=context, data=data)
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 865, in async_create_flow
    flow = handler()
TypeError: ABBPowerOnePVISunSpecConfigFlow.__init__() missing 1 required positional argument: 'config_entry'

alexdelprete, 2023-01-10:
Please try installing latest v2.3.1 and let me know if it fixes the issue. Thanks.

chrisvor, 2023-01-10:
yeap! now it works :) Thanks!!