alexdelprete/ha-abb-powerone-pvi-sunspec #50: Conflict with Modbus: official integration

alexdelprete, 2023-01-23:
Hello Alex,

I installed the new version after updating HA Core. 
HA receives data from my 2 inverters, the model is now recognised but LOG is showing the folowing error regarding modbus:

2023-01-23 11:00:41.597 ERROR (MainThread) [homeassistant.setup] Setup failed for modbus: Unable to import component: No module named 'pymodbus.client.sync'

Also, another device which uses modbus does not work, seemingly due to some sort of conflict.

Could you please help me in solving this?
Thanks again for your great support,
Domenico

_Originally posted by @usrpend in https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/issues/44#issuecomment-1400094365_

alexdelprete, 2023-01-23:
Domenico, there's another user @Polvere79 that had the same issue, I'll tag him because maybe he found out a solution to the conflict, without disabling modbus native.

Read here: https://community.home-assistant.io/t/custom-component-abb-power-one-fimer-pv-inverters-sunspec-modbus-tcp/316363/196?u=alexdelprete

alexdelprete, 2023-01-23:
Thanks for answering.

I have to setup modbus: again to understand what's going on...I used them both together for a while...but maybe something changed in latest HA versions.

twarnick, 2023-01-25:
Hi, I have updated ha core. Now I have the same error and the items for all my powermeters no longer available.

twarnick, 2023-01-25:
I just see, that I am not in ha core support.
I do not use your modul.
Sorry.

alexdelprete, 2023-01-26:
@Polvere79 @usrpend @twarnick I just release v2.5.0 with a new functionality and possibly the resolution of the conflict: HA modbus integration was changed pymodbus requirement to 3.1.1, so I did the same in the component. I hope that was the cause of the conflict, but I need you to test it and let me know.

Please also let me know if you're able to change configuration options like in the small animation in the changelog: https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/releases/tag/v2.5.0

Thanks.

usrpend, 2023-01-28:
Hello Alex,
thanks for your support. I installed v2.5.0 and the conflict is still in place. I do confirm I'm able to change the settings like in the small animation in the changelog. I will try to switch to DEV version of HA and will let you know about the results.

usrpend, 2023-01-28:
Update...
**I installed HA v.2023.2.0b3 and the conflict is now SOLVED.**
Anyway, I have another issue now: the details in the Energy panel are not available anymore at month/week/day/hour level stats. Only the totals (energy produced - energy to the net - energy from the net) are shown. It seems like a DB issue. How is this possible?

alexdelprete, 2023-01-28:
> I installed HA v.2023.2.0b3 and the conflict is now SOLVED.

because in 2.0b3 the same pymodbus version as HA is used. So it's not solved. :)

> How is this possible?

because the unique_id has changed. When you check the history of the production sensor, do you see all data from the past? if not, it means the unique_id has changed.

alexdelprete, 2023-01-28:
Regarding the issue with the Energy integration, show me a screenshot of what is missing...

alexdelprete, 2023-01-28:
> Hello Alex, thanks for your support. I installed v2.5.0 and the conflict is still in place. I do confirm I'm able to change the settings like in the small animation in the changelog. I will try to switch to DEV version of HA and will let you know about the results.

Don't do it in production, do it in a test environment, possibly.

usrpend, 2023-01-29:
Alex, don't know exactly why, but I just installed HA v.2023.2.0b4 and your version 2.5.0 and all seems OK, included energy stats:
as you can see in the attached screenshots, the panel 'Impianto Fotovoltaico' is retrieving data via stock Modbus under config.yaml, whereas the panels 'Fimer Inverter #1 and #2' on the right (your addon) are also working and retrieving data OK.
Also, the current and historical data in the Energy panel (picture 2 ) are showing normally.
The new button 'Configuration' in your new version is also working OK.
I can't figure out what is really changed in HA v. 2023.2.0b4 compared with v. 2023.2.0b3
 
[image]
[image]

alexdelprete, 2023-01-29:
> Alex, don't know exactly why, but I just installed HA v.2023.2.0b4 and your version 2.5.0 and all seems OK

What do you mean you don't know why? You installed HA beta. :)

As I explained before: HA Modbus integration has upgraded pymodbus component to the same version I'm using in my component, so there's no conflict anymore.

Happy everything's good now.

alexdelprete, 2023-02-02:
HA 2023.2.0 is finally released, and it includes pymodbus 3.1.1, same version as this component.

I just released [v2.6.0](https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/releases/tag/v2.6.0) that finally solves the conflict.

Thanks for the feedbacks.

closed 2023-02-02 by alexdelprete