alexdelprete/ha-abb-powerone-pvi-sunspec #54: Model unknown

asvel78, 2023-02-07:
Hi Alessandro, 

I receive this message in the logs

Model unknown, report to @alexdelprete on the forum the following data: Manuf.: ABB - Model: -3P38- - Options: 0x0BED/0xFF69/ - OptModel: 0x0BED/0xFF69/

my Model: UNO-DM-2.0-TL-PLUS

Regards.
LN

alexdelprete, 2023-02-07:
Hi,

there's a missing piece (`OptModelInt`) at the end of the logged line, should be something like this:

`Options: 0x10ED/0xFF69/ - OptModel: 0x10ED/0xFF69/ - OptModelInt: 16`

Could you paste the complete line here? I will add your model to the source, and in next release your model# will be correctly reported.

Thanks.

alexdelprete, 2023-02-07:
No problem, it happens. :)

I just released v2.6.1, thanks for your report.

If you now see the correct model in the sensor, please close the issue.

asvel78, 2023-02-13:
Solved, thanks Alex

closed 2023-02-13 by asvel78