alexdelprete/ha-abb-powerone-pvi-sunspec #66: Model unknown

naamah75, 2023-06-05:
I would like to report an unsupported inverter model:

```
(opt_comm_model) Model unknown, report to @alexdelprete on the forum the following data: Manuf.: Power-One - Model: -3N41- - Options: 0xF2 - OptModel: 0xF2 - OptModelInt: 242
(read_inv) Reading data failed! Please check Slave ID: 247
(read_inv) Reading data failed! Please check Reg. Base Address: 0
(read_data) Error: check Slave ID: 247 & Base Address: 0
```

alexdelprete, 2023-06-05:
The problem is not the unknown model, it's that you are not reading any data, ("reading data failed" in the log).

Check IP, port, Slave ID and Base address. Once you see data for the inverter, if the model # is unknown, then report back and we add it. It's just to have the right model written in the sensor, data must always be read.

naamah75, 2023-06-05:
your integration is working fine, but this message has been appearing in the logs for some time and therefore I decided to send you the feedback as written.

alexdelprete, 2023-06-06:
Ok, good to know, I saw those errors in the log I thought you couldn't read data. Sorry about the misunderstanding.

So I will add `OptModelInt: 242` but I need to know the name of the model to associate the appropriate string.

naamah75, 2023-06-06:
UNO-3.0-TL-OUTD

alexdelprete, 2023-06-06:
Thanks, added it. When I'll publish a new release you should see the correct model in the sensor and in the device info.

closed 2023-09-09 by alexdelprete