alexdelprete/ha-abb-powerone-pvi-sunspec #67: Model unkmown

Matz88, 2023-06-11:
Hi, I have a Uno DM 6.0 and I get the following error for unknown model:

Manuf.: ABB - Model: -3P31- - Options: 0x05ED/0xFF45/ - OptModel: 0x05ED/0xFF45/ - OptModelInt: 5

Some of the values are correct but mostly not with base 40000 and slave ID 1

What could be done?

alexdelprete, 2023-06-11:
The unknown model is just a cosmetic thing just to show the correct model (I don't know all of them, I'm building the model structure based on user feedbacks).

As for the values, show me a screenshot of all the sensors to see which ones are now working.

Matz88, 2023-06-11:
thanks for the feedback.
about the values, oddly after few hours I started getting reasonable values, see the temperature below:
[image]

alexdelprete, 2023-06-12:
that's the only value that you see strange? You said "mostly not correct" in the OP. :)

Also, as I wrote before, please provide the exact inverter model, otherwise I cannot include yours.

alexdelprete, 2023-06-12:
Can you please show a screenshot with all the sensors entities?

Are the values still wrong right now? I never happen to have a bad reading. Either it reads them, or it doesn't read them.

For MPPT string values, it can happen that the register map used by the component doesn't match with some inverters...but I have to check all the sensors before telling you what it could be.

alexdelprete, 2023-06-14:
Do you have one string or more than one?

Matz88, 2023-06-14:
One

Matz88, 2023-06-26:
Maybe I understood it. Could it be a calibration error? Looks like the temperature is jumping to a wrong value above 50°C namely 5.1 instead of 51, 5.2 instead of 52...just a speculation, see screenshot
[image]

closed 2023-09-09 by alexdelprete