alexdelprete/ha-abb-powerone-pvi-sunspec #69: Not getting DC value

ivanfmartinez, 2023-08-10:
I just installed this component v2.6.6 and found that I can see only a single DC Voltage (and Current) with value of -0.0 V. The AC and Energy values  appear to be OK.

```
2023-08-10 16:52:05.155 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] (read_inv) Manufacturer: Power-One
2023-08-10 16:52:05.155 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] (read_inv) Model: -3G03-
2023-08-10 16:52:05.155 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] (read_inv) Options: 6
2023-08-10 16:52:05.155 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] (opt_printable) opt_model: 6 - opt_model_int: 54
2023-08-10 16:52:05.155 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] (opt_comm_model) comm_model: PVI-6000-OUTD
2023-08-10 16:52:05.155 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] (read_inv) Version: C033

```

My inverter is single-phase version but running connected to two-phase AC (AC Voltage and Current are ok) with two DC MPPT strings .

@alexdelprete I have made a test a test branch (based on v2.6.6) to try to make it define single or multiple mppt sensors.

Can you check to see if you agree with this idea : https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/compare/v2.6.6...ivanfmartinez:ha-abb-powerone-pvi-sunspec:check_mppt

I know that there are a multi_mppt_nr on the code but I have tried to make just little changes now to make it works in my system with v2.6.6, as I have seen that the master branch have a lot of other commits.

alexdelprete, 2023-08-11:
Hi Ivan,

I noticed this issue being reported by a couple of users.

I'm in vacation for 2 weeks, will check your PR when I get back, thanks a lot for the contribution. I'm not a developer (last time I coded was 20y ago) but since I use HA I started again but for me everything's new so I mainly tinker to make stuff work as good as I can.

Will give you a feedback as soon as I'm back home.

Alessandro

ivanfmartinez, 2023-10-23:
Hi @alexdelprete ,

i have just update my HA instance and updated my changes to be compatible with your latest changes

https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/compare/v3.1.5...ivanfmartinez:ha-abb-powerone-pvi-sunspec:multi_mppt_2023_10

If you have time now please give the feedback

alexdelprete, 2023-10-25:
Did you make a PR?

ivanfmartinez, 2023-10-25:
> Did you make a PR?

Not yet, just send the diff here before to check what you think.
But I can make the PR if you agree or make the PR and discuss there.

Is my first time with this project and I did not know how you prefer to work.

alexdelprete, 2023-10-25:
> But I can make the PR if you agree or make the PR and discuss there.

Yes, prepare the PR so we can discuss there. Take into account that I'm not a developer (I was 20+ years ago), and I'm not an expert on GitHub etc., I'm learning but if you can contribute and help me I am more than open to receive support. :)

Thanks.

closed 2023-10-28 by ivanfmartinez