alexdelprete/ha-abb-powerone-pvi-sunspec #70: Integration Error after HA Update

marcolinux46, 2023-09-08:
Hi @alexdelprete ,

after last HA update i have the following error on log window about the integration.

`Logger: custom_components.abb_powerone_pvi_sunspec
Source: custom_components/abb_powerone_pvi_sunspec/__init__.py:228
Integration: ABB Power-One PVI SunSpec ([documentation](https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec), [issues](https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/issues))
First occurred: 05:32:57 (1 occurrences)
Last logged: 05:32:57

(read_data) Error: check Slave ID: 1 & Base Address: 40000`

Until this update everything worked fine

Can you help me to resolve this issue?

naamah75, 2023-09-08:
Same problem here

alexdelprete, 2023-09-08:
I upgraded to 2023.9 and I don't see this error.

woodrook, 2023-09-08:
I just tryed to remove the integration and reinstall with the same settings..........it show "no device or entities"

Before the 2023.9 upgrade everything was perferct

alexdelprete, 2023-09-08:
[image]
[image]

marcolinux46, 2023-09-08:
[image]

NicolasFOVET, 2023-09-08:
Hi, same issue for me, integration indicates inverter not available since I upgraded to 2023.9.0.

woodrook, 2023-09-08:
Solved! I’ve manually deleted abb integration and manually installed a new one. Configuration has been made from the mobile phone app instead od a computer browser.
Now everything seems to work. Keep finger crossed

marcolinux46, 2023-09-08:
I did the same but nothing change
Now i have no device or entity

marcolinux46, 2023-09-08:
Reconfigured now.... 
Same error

alexdelprete, 2023-09-08:
Nothing changed in 2023.9.0 that impacts the integration (pymodbus etc.). It's working fine here.

Make sure the IP is reachable (the VSN card) and that slave id and base address are ok.

marcolinux46, 2023-09-08:
@alexdelprete 
The ip is reachable and i checked the slave id and base address directly on inverter
Everything is ok
Nothing working after HA update
Before update the integration was good always perfect

marcolinux46, 2023-09-08:
I also test the TCP modbus with modpoll client
And i see all the data

marcolinux46, 2023-09-08:
I have also the Cnje/sunspec integration. 
Same issue... 
I left and reconfigured it and now is working

alexdelprete, 2023-09-08:
> I left and reconfigured it and now is working

what is working?

ANd also, tell me, what's this stuff highlighted in the red box?

[image]

marcolinux46, 2023-09-08:
@alexdelprete 

Now is working the Cjne/ha-sunspec integration. 
Your integration is still off. 

In the red box there are two consumption control but now are disabled.

marcolinux46, 2023-09-08:
Can i do some checks from ha?

alexdelprete, 2023-09-08:
> In the red box there are two consumption control but now are disabled.

So you modified the code? Reinstall the component from scratch. And also, post the entire HA log so we can check if you have other errors too.

marcolinux46, 2023-09-08:
No i don't modify the code
They are simple Riemann helpers
I try to reinstall from scratch

alexdelprete, 2023-09-08:
> I also test the TCP modbus with modpoll client And i see all the data

from where are you testing TCP modbus? how is HA installed? docker...bare metal...you need to test TCP modbus connection from the same instance HA is running, not only from your pc.

marcolinux46, 2023-09-08:
HA is installed on a VM
I can make some tests from terminal

alexdelprete, 2023-09-08:
> HA is installed on a VM
> I can make some tests from terminal

from the VM, check basic ping first, then Modbus TCP. The error is clear: it can't contact the VSN card.

marcolinux46, 2023-09-08:
Basic ping is working
How can i test modbus TCP?

marcolinux46, 2023-09-08:
I installed integration from scratch again 
Seems to be no errore occured

alexdelprete, 2023-09-08:
> I installed integration from scratch again Seems to be no errore occured

So it's working?

marcolinux46, 2023-09-08:
Nop

marcolinux46, 2023-09-08:
When i install the integration pack
But when i configure the inverter not working

alexdelprete, 2023-09-08:
setup a test HA instance from scratch (if you have a docker VM), and install from there to test.

marcolinux46, 2023-09-08:
Probably i understand the problem. 
Inside the integration folder is missing some files. 
Later i will install the package manually and i will let you know

alexdelprete, 2023-09-08:
> Probably i understand the problem. Inside the integration folder is missing some files. Later i will install the package manually and i will let you know

If you reinstalled from scratch, how can there be missing files?

Also, try the "redownload" option of HACS: open the component page from HACS and in the upper right, click the three dots and redownload. It will install all the files again.

If you're missing files, you have something not working correctly.

marcolinux46, 2023-09-08:
I removed the integration from HACS and i reinstalled it
This is what i see
[image]

Now i try to upload all the files and i let you know

marcolinux46, 2023-09-08:
Resolved.....
Files missing on integration folder after update
Problably something wrong with HACS

Thank you all for support

closed 2023-09-08 by alexdelprete