alexdelprete/ha-abb-powerone-pvi-sunspec #73: Update Problem (file missing)

marcolinux46, 2023-09-09:
@alexdelprete 

there's something wrong with HACS update process.
Today i update your integration to 2.6.7 via HACS and it is not working again.

This is the folder before update

[image]

and this is after update

[image]

Now i try to reload the file manually as yesterday

Thank you for your support

marcolinux46, 2023-09-09:
Integration Manually Installed

[image]

alexdelprete, 2023-09-09:
You are confusing things: there's a master branch

[image]

And there's the official release branch, tagged with the version (v2.6.7):

[image]

Don't upload manually if you don't understand that there are different branches, and always use the tagged release version. The master branch is my dev branch, and I experiment things, etc.

Use HACS when redownloading, and don't choose master from there, only choose the numbered versions.

marcolinux46, 2023-09-09:
@alexdelprete 

i chose the numbered release from HACS and now i have this file on integration folder

[image]

when i try to configure my inverter i receive this error on log again
(This issue is starting from 2023.9.0 HA update)

"Logger: custom_components.abb_powerone_pvi_sunspec
Source: custom_components/abb_powerone_pvi_sunspec/__init__.py:228
Integration: ABB Power-One PVI SunSpec (documentation, issues)
First occurred: 17:59:20 (3 occurrences)
Last logged: 17:59:33

(read_data) Error: check Slave ID: 1 & Base Address: 40000"

Thanks

alexdelprete, 2023-09-09:
> i chose the numbered release from HACS and now i have this file on integration folder

Ciao Marco, I found the problem, see release notes of latest v2.6.8 please.

Let me know if 2.6.8 works for you, thanks.

marcolinux46, 2023-09-09:
Ciao Alex, 

Tomorrow morning i will do some tests and i will let you know

Thanks

alexdelprete, 2023-09-09:
I managed to do one test right before the inverter shut off...only 2 minutes unfortunately. And it worked. Let me know...

marcolinux46, 2023-09-10:
Here i am,

1 - Inverter Up and Running (PING and NMAP OK)
2 - Removed all the old integration instances from HACS
3 - Installed from scratch, through HACS, version 2.6.8 integration package directly
4 - Installed the inverter --- Not Working ---

below are the results

Files on folder

[image]

Manifest.json

[image]

Service and Device

[image]

System LOG

Logger: custom_components.abb_powerone_pvi_sunspec
Source: custom_components/abb_powerone_pvi_sunspec/__init__.py:228
Integration: ABB Power-One PVI SunSpec ([documentation](https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec), [issues](https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/issues))
First occurred: 07:24:53 (3 occurrences)
Last logged: 07:25:06

(read_data) Error: check Slave ID: 1 & Base Address: 40000

Is there anything I can do to help you?

Thanks

naamah75, 2023-09-10:
Same issue as @marcolinux46

alexdelprete, 2023-09-10:
> Same issue as @marcolinux46

Thanks for the updates guys: I'm in contact with other devs of other inverter's components, they're all having problems with latest pymodbus versions, it seems there's more than one problem. The one I caught yesterday was blocking, but it seems there's a problem with the connect() function.

From what I read, HA 2023.9.2 will be released soon with a new pymodbus version (3.6 I guess) which should contain some fixes.

Stay tuned...

alexdelprete, 2023-09-10:
@marcolinux46 @naamah75 Guys, can you please try the master branch? You can redownload from HACS, and when the list of versions comes up, scroll down to the bottom and you'll see `master`.

Double-check that in the component's folder you have the following files, then restart and let me know how it goes. Thanks.

[image]

marcolinux46, 2023-09-10:
Here i am,

this is the version....same as yours:

[image]

this is the content integration folder:

[image]

And this is the final result:

[image]

Yesssssssssssssssssssssssssss

Good job @alexdelprete

alexdelprete, 2023-09-10:
That's awesome! :)

Thanks Marco. I didn't want to release this version because it's totally different and was a WIP since a couple of months. But I guess I'll have to introduce it sooner than I expected. It's still not tested properly yet...let's cross fingers. :)

marcolinux46, 2023-09-10:
No problem Alex,

I check the server every day, if I find any problems I will let you know.
For the moment i don't see the right model yet, but i thing thi is because is master branch.

Thank you for your support

alexdelprete, 2023-09-10:
> I check the server every day, if I find any problems I will let you know. For the moment i don't see the right model yet, but i thing thi is because is master branch.

Damn it, I forgot your model and I just released v3.0.0. :(

When you install v3.0.0, modify const.py, lines 204-205:

```python
DEVICE_MODEL = {
      0: "UNO-DM-3.3-TL-PLUS",
```

Basically you need to add the 0: line. Then wait for v3.0.1.

marcolinux46, 2023-09-10:
Ahahahahahaha no worry about it,

Update to 3.0.0 Major release and it seems working fine and device model too.

[image]

Good job @alexdelprete 

I will update during this period if some issues occur.

Have a good weekend

closed 2023-09-10 by alexdelprete

alexdelprete, 2023-09-11:
> Same issue as @marcolinux46

@naamah75 did you try v3.0.0? Is it working for you?

naamah75, 2023-09-12:
> 

Yes! it work! thanks!

alexdelprete, 2023-09-15:
@marcolinux46 @naamah75 please test beta.5. I'm testing it since 24h without disabling the integration or any external automation. The solution was to test if Modbus TCP was available at socket level before trying the modbus connect (pymodbus call). This avoids all the errors etc. The retry mechanism is managed by HA itself, and the integration will be online automatically as soon as the TCP port is available (inverter on) and the Modbus TCP connect is processed correctly.

marcolinux46, 2023-09-16:
@alexdelprete i installed the beta.5 and everythink is working fine.
No problem for now.

naamah75, 2023-09-16:
@alexdelprete ... installed b5, now it works. I will keep an eye on it during the day.

alexdelprete, 2023-09-16:
I tested it for 2 days, it's working fine here when inverter powers off and on, etc. No log spamming either. I released v3.1.0 final based on this.

Let me know how it goes...thanks for the help.

marcolinux46, 2023-09-16:
@alexdelprete installed now 3.1.0
No problem during installation and no error or warning on log. 

If something happens i will let you know

Thanks

alexdelprete, 2023-09-16:
> installed now 3.1.0

Just released 3.1.1 with a minor improvement. But it shouldn't affect you, install it when you can.