alexdelprete/ha-abb-powerone-pvi-sunspec #86: The inverter disconnects and reconnects frequently.

Pippo1989, 2023-10-17:
Hi. For a few weeks now, my inverter has started disconnecting and reconnecting frequently and the sensors become, consequently, unavailable.

**Device specifications:**
UNO-DM-3.0-TL-PLUS
ABB
Firmware: 2201A

**Inverter connection options:**
TCP port: 502
Modbus Slave address of the inverter: 254
Modbus Register Map Base Address: 40000
Polling period: 30

**Hardware**: Raspberry Pi4 with Hassio

You can find attached the complete logs from the restart of Home Assistant and a screenshot of a graphic showing the frequent disconnections.

If it can be useful, I've noticed that, among the logs, you can find lines like this:

2023-10-17 10:28:35.628 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] ABB SunSpec Update data coordinator update
2023-10-17 10:28:35.628 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Hub connect to IP: 192.168.1.130 port: 502 slave id: 254 timeout: 29
2023-10-17 10:28:35.628 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Check_Port: opening socket on 192.168.1.130:502 with a 3.0s timeout.
2023-10-17 10:28:38.633 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Check_Port (ERROR): port not available on 192.168.1.130:502 - error: 11
2023-10-17 10:28:38.634 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Inverter not ready for Modbus TCP connection
2023-10-17 10:28:38.634 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Async Update Data error: Inverter not active on 192.168.1.130:502
2023-10-17 10:28:38.635 ERROR (MainThread) [custom_components.abb_powerone_pvi_sunspec] Error fetching abb_powerone_pvi_sunspec data: 
2023-10-17 10:28:38.635 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Finished fetching abb_powerone_pvi_sunspec data in 3.008 seconds (success: False)

Also, there wouldn't seem to be any problems using the inverter's official application (Energy Viewer), because it shows normal values.

Thanks in advance.

[home-assistant_abb_powerone_pvi_sunspec_2023-10-17T08-39-35.512Z.log](https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/files/12927065/home-assistant_abb_powerone_pvi_sunspec_2023-10-17T08-39-35.512Z.log)
[image]

alexdelprete, 2023-10-17:
The log is pretty clear:

```
2023-10-17 10:28:35.628 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Check_Port: opening socket on 192.168.1.130:502 with a 3.0s timeout.
2023-10-17 10:28:38.633 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Check_Port (ERROR): port not available on 192.168.1.130:502 - error: 11
2023-10-17 10:28:38.634 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Inverter not ready for Modbus TCP connection
```

That means the component couldn't open a socket on port 502 of the VSN card. It's a network issue that you need to check. Check wifi of the VSN card. Also, from HA container/host, do a tcp port check on inverter:502 every 30s to see if it's available.

Pippo1989, 2023-10-22:
Hi,

As suggested, I ran some tests, including checking the TCP connection, but I didn't get any results that could explain what caused the problem.

I also checked that the configuration of the wifi network through the access point generated by the inverter was carried out correctly.

On the other hand, the frequent disconnections appeared out of the blue, without any intervention on my part: therefore it is not even possible for me to make any hypotheses about the cause of the problem.

As a contribution, I tried to install the SunSpec (generic) integration using the same configuration. The result is that the sensors obtained (including AC/DC production) are not subject to disconnection, unlike this integration.

Thank you.

alexdelprete, 2023-10-22:
Can't tell you why a python code can't make a socket connection reliably in your case, if not because of wifi network issues.

The socket connection is a very basic test, if it fails, there's a network problem. That's not my opinion, it's a fact.

But I think you found the solution: use the Sunspec integration, it's an excellent alternative.

closed 2023-11-07 by alexdelprete