alexdelprete/ha-abb-powerone-pvi-sunspec #91: v3.2.0 introduces static values values for DCx Current, DCx Voltage, and DCx Power

jumpinf00l, 2023-11-26:
Since updating from v3.1.5 to v3.2.0, the entites for DCx Current, DCx Voltage, and DCx Power are set at 1, rather than the expected values for each DC array (two in my case, DC1 and DC2). The solar inverter is already updated to latest firmware version 2201A and RS485 protocol is set to Protocol Aurora Server, but this seems to make no difference to any of the values shown in this integration (I've tried all four options)

**Expected Result:**
DC1 Current, DC1 Voltage, DC1 Power, DC2 Current, DC2 Voltage, and DC2 Power entities show correct values for each DC array

**Current Result:** 
DC1 Current, DC1 Voltage, DC1 Power, DC2 Current, DC2 Voltage, and DC2 Power entities show value of 1

**Hardware:**
Make: ABB
Model: UNO-DM-5.0-TL-PLUS
Firmware: 2201A
Inverter Type: Single Phase

**Integration options:**
TCP port: 502
Modbus Slave address of the inverter: 254
Modbus Register Map Base Address: 40000
Polling period: 30

Thanks in advance, cheers for the great integration

[home-assistant_abb_powerone_pvi_sunspec_2023-11-26T02-39-48.861Z.log](https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/files/13465858/home-assistant_abb_powerone_pvi_sunspec_2023-11-26T02-39-48.861Z.log)
[image]

alexdelprete, 2023-11-26:
The root cause is this: https://community.home-assistant.io/t/custom-component-abb-power-one-fimer-pv-inverters-sunspec-modbus-tcp/316363/345?u=alexdelprete

The other strange thing is that the MPPT Count in your case is 303, and that's not a consistent value, you should have 2 there.

Unfortunately some inverter models have a different register map, and in order to resolve this I would need to add some exception code to manage these differences. I was not very inclined to do that...but probably I'll have to address these cases.

Can you use qmodbusmaster to show the contents of the registers?

I'll tag @ivanfmartinez because he has a monophase inverter like you with 2 MPPTs.

alexdelprete, 2023-11-26:
@jumpinf00l I found the specific offset for Model 160 (UNO-DM-PLUS and REACT2 inverters only), can you please install the master version in HACS and let me know if it works? A debug log and screenshot would be perfect, thanks.

In case you don't know how to install the `master` branch version: go to HACS, open my integration, hit the 3-dots on the upper right and click `Redownload`, then choose `master`:
[image]

jumpinf00l, 2023-11-27:
@alexdelprete Perfect, that seems to have sorted it! Thanks very much for looking at it

[home-assistant_abb_powerone_pvi_sunspec_2023-11-27T09-45-29.697Z.log](https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/files/13472589/home-assistant_abb_powerone_pvi_sunspec_2023-11-27T09-45-29.697Z.log)
[image]

Here's the screenshot from qmodmaster if it helps for future (I hope I've set it up right)
[image]

Cheers for the quick reply and fix!

alexdelprete, 2023-11-27:
Perfect, here you can see the fix in action:

```console
(read_rt_160) **_Model not 160 try another offset - multi_mppt_id: 120_**
(read_rt_160) Slave ID: 254
(read_rt_160) Base Address: 40000
(read_rt_160) **_Model is 160 (UNO-DM/REACT2) - multi_mppt_id: 160_**
(read_rt_160) mppt_nr 2
(read_rt_160) Completed
```

I'll close the issue if it's ok for you and release a new version.

Thanks for the help.

closed 2023-11-27 by alexdelprete

alexdelprete, 2023-11-27:
> Cheers for the quick reply and fix!

I noticed a strange thing from yous screenshot: DC Power is not the sum of DC1 Power + DC2 Power. Can you please monitor that when you have time and let me know if that's always the case? On my inverter it is the sum of the the DCx Power of the MPPTs.

Please let me know, thanks.

jumpinf00l, 2023-11-27:
Sure thing, I’ll monitor and let you know

ivanfmartinez, 2023-11-27:
> I noticed a strange thing from yous screenshot: DC Power is not the sum of DC1 Power + DC2 Power. Can you please monitor that when you have time and let me know if that's always the case? On my [...]

In my case DC power also is not the exactly the sum of DC1 and DC2, but are near. Probably this is related to different timings for the register updates.

alexdelprete, 2023-11-27:
> In my case DC power also is not the exactly the sum of DC1 and DC2, but are near. Probably this is related to different timings for the register updates.

Not with the integration, but it reads everything in page mode, in 0.1s.

This was the main reason I started developing it.

ivanfmartinez, 2023-11-27:
> > In my case DC power also is not the exactly the sum of DC1 and DC2, but are near. Probably this is related to different timings for the register updates.
> 
> Not with the integration, because it reads everything in page mode, in 0.1s.
> 
> This was the main reason I started developing it.

The difference can be already inside the inverter if the MPPTs are independent and they put an separate process to calculate total this will happen, and many other possibilities.

ivanfmartinez, 2023-11-27:
> That's not how the data logger works: it gathers all data via internal
> Aurora protocol, and it provides new modbus data every 60s (if I remember
> correctly, might be even longer). And I wasn't able to change the default
> period.
>
> So there's no discrepancy.

For me since beginning there was this difference most of the time.
Maybe my model works differently than yours.

alexdelprete, 2023-11-27:
> For me since beginning there was this difference most of the time. Maybe my model works differently than yours.
> […](#)

The VSN cards work all the same. I think it might be something specific to some models, or firmware of those models, maybe the DC Power takes into account power factor or something else, I don't know. The difference is minimal I guess. How much in your case?

alexdelprete, 2023-11-27:
> It's really a minimal difference.

so I guess nothing to worry about. If you notice wide gaps, please let me know. I know we can't do much about it, since that's what the inverter is reporting, but if some users report this, at least we know about it.

Thanks for the help.

jumpinf00l, 2023-11-28:
> so I guess nothing to worry about. 

It hasn’t been more than 20W difference since I’ve been monitoring, very negligible and not breaking anything

alexdelprete, 2023-11-28:
Thank you for the feedbacks gentlemen. Another improvement in the integration, happy about that.

Now next big step will be to pull data from VSNx00 card via HTTP and parse json. This will allow the integration to have ALL data, including battery, meters, etc. without worrying about registers, address maps differences among models, etc.

@ivanfmartinez are you a developer? would you be interested in helping with that? I already managed to implement a python script that authenticates with VSN300 and VSN700 (they have different auth schemes) and gathers all JSON.

alexdelprete, 2023-11-28:
Oh my god, @ivanfmartinez, I just found out checking my repo [here](https://github.com/alexdelprete/vsnx00-monitor) that you were a contributor in 2016 to the script I started from for the VSN300. We really live in a small world...😄 

Anyway, in that repo you'll find a script that is tested on 3 inverter types, one is mine with the VSN300 and then other 2 people tested it and they have VSN700 cards. I put sample data in folder `data-samples`, if you want to take a look at it.

Let me know, it would be fantastic to have an integration that works both with Modbus TCP and HTTP/JSON, and the user can configure it as they like.

Let me know what you think about it...thanks.

ivanfmartinez, 2023-11-28:
> OMG, @ivanfmartinez, I just found out checking my repo [here](https://github.com/alexdelprete/vsnx00-monitor) that you were a contributor in 2016 to the script I started from for the VSN300. We really live in a small world...😄
> 
> Anyway, in that repo you'll find a script that is tested on 3 inverter types, one is mine with the VSN300 and then other 2 people tested it and they have VSN700 cards. I put sample data in folder `data-samples`, if you want to take a look at it.
> 
> It would be fantastic to have an integration that works both with Modbus TCP and HTTP/JSON, and the user can configure it as they like.
> 
> Let me know what you think about it...thanks.

I'm a developer, but don´t have much time available, but I can help. 

When I installed my system in 2016 I have looked many options to integrate with my openhab and pvoutput. The world becomes smaller with this public repositories.

I have tested your script in my inverter and committed a new file in my fork, my json is different from the other two files already in repository.