alexdelprete/ha-abb-powerone-pvi-sunspec #95: Integrazione di un inverete PVI6000 del 2010

giobbe70, 2023-12-07:
Salve,
vorrei integrare il mio inverte PVI6000 del 2010 in HA.
Ho contattato PoverOne e mi hanno detto che VSN300 non è compatibile con i vecchi inverter ma ci vuole il VSN700 che costa 700€!
C'e' qualche modo per usare il VSN300 perche non si può spendere 700€ per un logger!
Grazie

alexdelprete, 2023-12-07:
VSN300 è compatibile con il tuo inverter, stando al [documento di compatibilità ufficiale FIMER](https://www.fimer.com/sites/default/files/FIMER_Accessoriescompatibilitymatrix_ApplicationNote_EN.-RevC.pdf):

[image]

giobbe70, 2023-12-07:
Il mio inverter è del 2010 loro dicono che non è compatibile...non vorrei spendere soldi a vuoto.

[image]

[image]

giobbe70, 2023-12-07:
Secondo te è compatibile, lo posso comprare?

giobbe70, 2023-12-07:
La porta per collegarla pare che ci sia...

[image]

alexdelprete, 2023-12-07:
Hai letto il mio messaggio precedente? è UFFICIALMENTE compatibile.

giobbe70, 2023-12-07:
corretto post