alexdelprete/ha-abb-powerone-pvi-sunspec #98: UNO-DM-4.0-TL-PLUS 140055-3P22-1721

cardil1970, 2024-01-04:
Ciao Alex,
ho scaricato la tua integrazione per provare ad integrare l'inverter in oggetto, ma non ci riesco...ti chiedo cortesemente supporto.
Il modello è nell'oggetto: Questo il log che ho scaricato dal debug:
2024-01-04 15:24:10.602 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Setup config entry for ABB
2024-01-04 15:24:10.602 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Data: {'port': 502, 'slave_id': 1, 'base_addr': 40084, 'scan_interval': 31, 'name': 'ABB Inverter', 'host': '192.168.1.36'}
2024-01-04 15:24:10.603 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Options: {}
2024-01-04 15:24:10.603 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Setup entry with scan interval 0:00:31. Host: 192.168.1.36 Port: 502 ID: 1
2024-01-04 15:24:10.603 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] ABB SunSpec Update data coordinator update
2024-01-04 15:24:10.603 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Hub connect to IP: 192.168.1.36 port: 502 slave id: 1 timeout: 30
2024-01-04 15:24:10.603 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Check_Port: opening socket on 192.168.1.36:502 with a 3.0s timeout.
2024-01-04 15:24:10.610 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Check_Port (SUCCESS): port open on 192.168.1.36:502
2024-01-04 15:24:10.610 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Inverter ready for Modbus TCP connection
2024-01-04 15:24:10.621 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Modbus TCP Client connected
2024-01-04 15:24:10.621 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Start Get data (Slave ID: 1 - Base Address: 40084)
2024-01-04 15:24:10.682 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] (read_rt_1) Slave ID: 1
2024-01-04 15:24:10.682 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] (read_rt_1) Base Address: 40084
2024-01-04 15:24:10.682 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Async Update Data error: 'ascii' codec can't decode byte 0xff in position 10: ordinal not in range(128)
2024-01-04 15:24:10.682 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Finished fetching abb_powerone_pvi_sunspec data in 0.079 seconds (success: False)

Se hai pazienza di darmi qualche dritta, te ne sarei grato.

PS: Ho già aperto un thread sull'argomento, ma è andato perso perchè ho avuto altre difficoltà che mi hanno distolto dal seguire la mia richiesta.
Grazie Comunque,
C.

alexdelprete, 2024-01-04:
Perché hai usato base address 40084? Di solita è 40000, hai già provato?

Riprova con 40000 e fammi vedere il debug. Inoltre, sei sicuro lo slave id sia 1?

cardil1970, 2024-01-05:
Ciao Alex, avevo usato la 40000, ma non avevo avuto successo, così googolando, avevo trovato altre configurazioni e c'era il valore 40084. Oggi ho reimpostato la 40000 e sembra funzionare. Per lo slave id =1, idem. 
Se mi volessi dire come si idividuano questi valori, seppur avendoli indovinati nella mia configurazione, potrei imparare qualcosa x il futuro. Non ho accesso al pannello avanzato col fimer perchè va richiesta una pwd aggiuntiva e temporanea ed al momento non devo fare configurazioni avanzate.
Mi scuso per averti disturbato x la seconda volta, ma ho seguito altri post nelle tue discussions e ho reinstallato il tuo software risolvendo il mio piccolo problema.
Grazie per il  momento.

alexdelprete, 2024-01-05:
Le istruzioni sono testate e chiare, quando c'è un problema invece di smanettare e perdere ore è meglio chiedere ed aspettare risposta. :)

Puoi verificare i parametri con un client modbus, tipo qmodmaster o modpoll. Per lo slave id puoi verificarlo direttamente sul pannello dell'inverter, oppure su Aurora Vision, l'app sul cloud per il monitoraggio:

[image]

cardil1970, 2024-01-05:
Hai perfettamente ragione! Ti avevo scritto, ed ovviamente tu avevi risposto, ma sono stato sopraffatto da altri eventi e mi sono perso.
Ti ringrazio per il tuo prezioso supporto.
C.

alexdelprete, 2024-01-05:
Di nulla.

Ciao,

Alessandro

closed 2024-01-05 by alexdelprete