alexdelprete/ha-abb-powerone-pvi-sunspec #162: Unload failing

woodrook, 2024-07-25:
### System Health details

## System Information

version | core-2024.7.3
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 4.4.302+
arch | x86_64
timezone | Europe/Rome
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1383
Downloaded Repositories | 19

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 16
views | 2
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | July 16, 2024 at 10:10 AM
-- | --
current_recorder_run | July 24, 2024 at 10:08 AM
estimated_db_size | 558.72 MiB
database_engine | sqlite
database_version | 3.45.3

</details>

### Checklist

- [X] I have enabled debug logging for my installation.
- [X] I have filled out the issue template to the best of my ability.
- [X] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [X] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Every morning integration show the message "Failed to unload" and i need to restart HA to let the inverter integraton works again.

I have spook that unload the integration when the inverter goes off in the evening and reload it when the inverter restart in the morning

Debug log show many things but (in my own opinion) i think the problem could be the "unloading" process that recently doesn't work.

Thanks, Gianluca

### Reproduction steps

unload integration with spook

### Debug logs

```text
2024-07-24 21:08:01.977 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry ABB Inverter for abb_powerone_pvi_sunspec
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 801, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/abb_powerone_pvi_sunspec/__init__.py", line 146, in async_unload_entry
    hass.data[DOMAIN][config_entry.entry_id].update_listener()
    ~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^
KeyError: '68e100a5b589d468733b76141e936def'
```

closed 2024-07-25 by woodrook

alexdelprete, 2024-07-25:
You should not unload the integration in the evening, leave it working, you will have a warning when the inverter goes offline, then HA retries to reconnect quietly, without any entry in the log, since I'm using DataCoordinator that handles this scenario automatically.

Have you tried "disabling" the integration from HA? does it give you an error?

woodrook, 2024-07-25:
Ok Alex, i've removed spook automation, let's see.......thank you for your reply.

Manual disabling require a HA reboot to completely disable ABB integration. After that the integration could be enabled without any problem.

Thanks

alexdelprete, 2024-07-25:
Nope. I used Spook in the past to enable/disable the integration on the fly, before I implemented DataCoordinator, because it spammed the log. No need to restart HA.

This was the automation to disable/enable it on VSN300 disconnect:

```yaml
alias: State of ABB Custom Component
description: >-
  Ensure ABB Component is running in the morning and disable it at sunset to
  avoid filling log with errors
trigger:
  - platform: state
    id: trigger_abb_vsn300_connected
    entity_id:
      - binary_sensor.ping_sensor_abb_vsn300
    from: "off"
    to: "on"
    for:
      hours: 0
      minutes: 0
      seconds: 20
  - platform: state
    id: trigger_abb_vsn300_disconnected
    entity_id:
      - binary_sensor.ping_sensor_abb_vsn300
    from: "on"
    to: "off"
    for:
      hours: 0
      minutes: 0
      seconds: 20
condition: []
action:
  - choose:
      - conditions:
          - condition: trigger
            id: trigger_abb_vsn300_connected
        sequence:
          - service: homeassistant.enable_config_entry
            data:
              config_entry_id: 54dcc9b0ac601ef0a7d78ebcd861cc76
          - service: homeassistant.reload_config_entry
            data: {}
            target:
              device_id: ab0bea32218dfd580d560e46fc85c722
      - conditions:
          - condition: trigger
            id: trigger_abb_vsn300_disconnected
        sequence:
          - service: homeassistant.disable_config_entry
            data:
              config_entry_id: 54dcc9b0ac601ef0a7d78ebcd861cc76
mode: single

```

Anyway, like I said, don't disable it, there's no need.