alexdelprete/ha-abb-powerone-pvi-sunspec #165: Failed with error: unpack requires a buffer of 2 bytes on v3.5.0

nonth, 2024-07-29:
### System Health details

## System Information

version | core-2024.7.3
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.12.4
os_name | Linux
os_version | 6.5.0-45-generic
arch | x86_64
timezone | Asia/Bangkok
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
GitHub API Calls Remaining | 5000
Installed Version | 1.34.0
Stage | running
Available Repositories | 1398
Downloaded Repositories | 15

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | January 1, 2018 at 7:00 AM
relayer_connected | false
relayer_region | null
remote_enabled | true
remote_connected | false
alexa_enabled | false
google_enabled | false
remote_server | null
certificate_status | null
instance_id | 39fe8ae6d4a44c54a5102898879b3151
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 7
-- | --
resources | 9
views | 13
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | July 20, 2024 at 12:25 PM
-- | --
current_recorder_run | July 29, 2024 at 8:35 PM
estimated_db_size | 401.80 MiB
database_engine | sqlite
[... 66 lines cut ...]
homeassistant  | 2024-07-29 18:30:45.772 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (opt_printable) opt_model: r - opt_model_int: 114
homeassistant  | 2024-07-29 18:30:45.772 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (opt_comm_model) comm_model: TRIO-5.8-OUTD
homeassistant  | 2024-07-29 18:30:45.772 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Version: CAA8
homeassistant  | 2024-07-29 18:30:45.772 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Sernum: 108624-3M97-5019
homeassistant  | 2024-07-29 18:30:45.779 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Slave ID: 247
homeassistant  | 2024-07-29 18:30:45.779 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Base Address: 0
homeassistant  | 2024-07-29 18:30:45.779 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Inverter Type (int): 103
homeassistant  | 2024-07-29 18:30:45.780 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Inverter Type (str): Three Phase
homeassistant  | 2024-07-29 18:30:45.780 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Total Energy Value Read: 25487796
homeassistant  | 2024-07-29 18:30:45.780 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Total Energy Previous Value: 1
homeassistant  | 2024-07-29 18:30:45.780 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) DC Power Value read: 0
homeassistant  | 2024-07-29 18:30:45.780 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Temp Oth Value read: 37.1
homeassistant  | 2024-07-29 18:30:45.782 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Temp Cab Value read: 37.3
homeassistant  | 2024-07-29 18:30:45.782 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Device Status Value read: Out OC
homeassistant  | 2024-07-29 18:30:45.782 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Status Vendor Value read: Run
homeassistant  | 2024-07-29 18:30:45.782 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Completed
homeassistant  | 2024-07-29 18:30:45.791 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_160) Slave ID: 247
homeassistant  | 2024-07-29 18:30:45.791 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_160) Base Address: 0
homeassistant  | 2024-07-29 18:30:45.791 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_160) Model not 160 try another offset - multi_mppt_id: 120
homeassistant  | 2024-07-29 18:30:45.802 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_160) Slave ID: 247
homeassistant  | 2024-07-29 18:30:45.802 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_160) Base Address: 0
homeassistant  | 2024-07-29 18:30:45.802 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] read_sunspec_modbus: failed with error: unpack requires a buffer of 2 bytes
homeassistant  | 2024-07-29 18:30:45.804 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Coordinator Update Error:  at 2024-07-29 18:30:45.738373
homeassistant  | 2024-07-29 18:30:45.804 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Finished fetching abb_powerone_pvi_sunspec (108624-3M97-5019) data in 0.066 seconds (success: False)
```

nonth, 2024-07-30:
Hi @alexdelprete ,

Thanks for an update but still getting the same error. Here's the log from v3.5.1-beta.1.

```
homeassistant  | -------------------------------------------------------------------
homeassistant  | ABB/Power-One/FIMER PVI SunSpec ModBus TCP
homeassistant  | Version: 3.0.0
homeassistant  | by @alexdelprete
homeassistant  | This is a custom integration!
homeassistant  | If you have any issues with this you need to open an issue here:
homeassistant  | https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/issues
homeassistant  | -------------------------------------------------------------------
homeassistant  | 
homeassistant  | 2024-07-30 10:42:06.892 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Setup config_entry for abb_powerone_pvi_sunspec
homeassistant  | 2024-07-30 10:42:06.892 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Scan Interval: scan_interval=60 update_interval=0:01:00
homeassistant  | 2024-07-30 10:42:06.892 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Coordinator Config Data: {'name': 'VSN300', 'host': '192.168.1.111', 'port': 502, 'slave_id': 247, 'base_addr': 0, 'scan_interval': 60}
homeassistant  | 2024-07-30 10:42:06.893 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Coordinator API init: Host: 192.168.1.111 Port: 502 ID: 247 ScanInterval: 60
homeassistant  | 2024-07-30 10:42:06.893 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Data Coordinator: Update started at 2024-07-30 10:42:06.893048
homeassistant  | 2024-07-30 10:42:06.893 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] API Client connect to IP: 192.168.1.111 port: 502 slave id: 247 timeout: 59
homeassistant  | 2024-07-30 10:42:06.893 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port: opening socket on 192.168.1.111:502 with a 3.0s timeout.
homeassistant  | 2024-07-30 10:42:07.234 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port (SUCCESS): port open on 192.168.1.111:502
homeassistant  | 2024-07-30 10:42:07.235 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Inverter ready for Modbus TCP connection
homeassistant  | 2024-07-30 10:42:07.247 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Modbus TCP Client connected
homeassistant  | 2024-07-30 10:42:07.248 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Start Get data (Slave ID: 247 - Base Address: 0)
homeassistant  | 2024-07-30 10:42:07.270 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Slave ID: 247
homeassistant  | 2024-07-30 10:42:07.271 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Base Address: 0
homeassistant  | 2024-07-30 10:42:07.271 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Manufacturer: Power-One
homeassistant  | 2024-07-30 10:42:07.271 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Model: -3M97-
homeassistant  | 2024-07-30 10:42:07.271 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Options: r
homeassistant  | 2024-07-30 10:42:07.271 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (opt_printable) opt_model: r - opt_model_int: 114
homeassistant  | 2024-07-30 10:42:07.271 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (opt_comm_model) comm_model: TRIO-5.8-OUTD
homeassistant  | 2024-07-30 10:42:07.271 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Version: CAA8
homeassistant  | 2024-07-30 10:42:07.271 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Sernum: 108624-3M97-5019
homeassistant  | 2024-07-30 10:42:07.304 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Slave ID: 247
homeassistant  | 2024-07-30 10:42:07.304 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Base Address: 0
homeassistant  | 2024-07-30 10:42:07.304 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Inverter Type (int): 103
homeassistant  | 2024-07-30 10:42:07.304 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Inverter Type (str): Three Phase
homeassistant  | 2024-07-30 10:42:07.304 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Total Energy Value Read: 25491184
homeassistant  | 2024-07-30 10:42:07.304 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Total Energy Previous Value: 1
homeassistant  | 2024-07-30 10:42:07.304 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) DC Power Value read: 1516
homeassistant  | 2024-07-30 10:42:07.305 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Temp Oth Value read: 42.4
homeassistant  | 2024-07-30 10:42:07.305 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Temp Cab Value read: 40.1
homeassistant  | 2024-07-30 10:42:07.305 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Device Status Value read: Out OC
[... cut ...]
homeassistant  | 2024-07-30 10:42:29.079 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] read_sunspec_modbus: failed with error: unpack requires a buffer of 2 bytes
homeassistant  | 2024-07-30 10:42:29.079 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Coordinator Update Error:  at 2024-07-30 10:42:28.307010
homeassistant  | 2024-07-30 10:42:29.079 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Finished fetching abb_powerone_pvi_sunspec (108624-3M97-5019) data in 0.772 seconds (success: False)
```

alexdelprete, 2024-07-30:
Strange. I used [this](https://view.officeapps.live.com/op/view.aspx?src=https%3A%2F%2Fsunspec.org%2Fwp-content%2Fuploads%2F2019%2F09%2FSunSpec-PICS-ABB-TRIO-50.0-TL-OUTD.xlsx&wdOrigin=BROWSELINK) register map (for the TRIO-50.0-TL-OUTD):

You can see that at address 40208 there's an int16 representing the model#, and it should be 160.

Can you check with qmodmaster reading from address 40200 if at 40208 you read the 160 value?

If it's not there, then the register map is not correct.

alexdelprete, 2024-07-30:
@nonth can you please test beta.2? I noticed I was decoding int16 type but it should be an uint16. I hope that was the issue. I also added some debug log to see where it's stopping, please post the debug log like you did before.

Thanks for the help.

nonth, 2024-07-31:
@alexdelprete  here's a log from beta 2, It looks like it stops around [line 549](https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/blob/master/custom_components/abb_powerone_pvi_sunspec/api.py#L549C9-L549C52). 

Anything I can do to help from my end?

Thanks!

```
homeassistant  | -------------------------------------------------------------------
homeassistant  | ABB/Power-One/FIMER PVI SunSpec ModBus TCP
homeassistant  | Version: 3.0.0
homeassistant  | by @alexdelprete
homeassistant  | This is a custom integration!
homeassistant  | If you have any issues with this you need to open an issue here:
homeassistant  | https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/issues
homeassistant  | -------------------------------------------------------------------
homeassistant  | 
homeassistant  | 2024-07-31 11:01:02.326 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Setup config_entry for abb_powerone_pvi_sunspec
homeassistant  | 2024-07-31 11:01:02.326 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Scan Interval: scan_interval=60 update_interval=0:01:00
homeassistant  | 2024-07-31 11:01:02.326 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Coordinator Config Data: {'name': 'VSN300', 'host': '192.168.1.111', 'port': 502, 'slave_id': 247, 'base_addr': 0, 'scan_interval': 60}
homeassistant  | 2024-07-31 11:01:02.326 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Coordinator API init: Host: 192.168.1.111 Port: 502 ID: 247 ScanInterval: 60
homeassistant  | 2024-07-31 11:01:02.326 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Data Coordinator: Update started at 2024-07-31 11:01:02.326924
homeassistant  | 2024-07-31 11:01:02.328 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] API Client connect to IP: 192.168.1.111 port: 502 slave id: 247 timeout: 59
homeassistant  | 2024-07-31 11:01:02.328 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port: opening socket on 192.168.1.111:502 with a 3.0s timeout.
homeassistant  | 2024-07-31 11:01:02.464 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port (SUCCESS): port open on 192.168.1.111:502
homeassistant  | 2024-07-31 11:01:02.464 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Inverter ready for Modbus TCP connection
homeassistant  | 2024-07-31 11:01:02.471 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Modbus TCP Client connected
homeassistant  | 2024-07-31 11:01:02.471 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Start Get data (Slave ID: 247 - Base Address: 0)
homeassistant  | 2024-07-31 11:01:02.486 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Slave ID: 247
homeassistant  | 2024-07-31 11:01:02.486 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Base Address: 0
homeassistant  | 2024-07-31 11:01:02.486 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Manufacturer: Power-One
homeassistant  | 2024-07-31 11:01:02.486 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Model: -3M97-
homeassistant  | 2024-07-31 11:01:02.486 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Options: r
homeassistant  | 2024-07-31 11:01:02.486 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (opt_printable) opt_model: r - opt_model_int: 114
homeassistant  | 2024-07-31 11:01:02.486 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (opt_comm_model) comm_model: TRIO-5.8-OUTD
homeassistant  | 2024-07-31 11:01:02.486 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Version: CAA8
homeassistant  | 2024-07-31 11:01:02.486 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Sernum: 108624-3M97-5019
homeassistant  | 2024-07-31 11:01:02.515 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Slave ID: 247
homeassistant  | 2024-07-31 11:01:02.515 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Base Address: 0
homeassistant  | 2024-07-31 11:01:02.515 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Inverter Type (int): 103
homeassistant  | 2024-07-31 11:01:02.515 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Inverter Type (str): Three Phase
homeassistant  | 2024-07-31 11:01:02.515 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Total Energy Value Read: 25504416
homeassistant  | 2024-07-31 11:01:02.515 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Total Energy Previous Value: 1
homeassistant  | 2024-07-31 11:01:02.515 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) DC Power Value read: 1768
homeassistant  | 2024-07-31 11:01:02.515 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Temp Oth Value read: 44.3
homeassistant  | 2024-07-31 11:01:02.516 DEBUG (SyncWorker_6) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Temp Cab Value read: 42.0
[... cut ...]
  "domain": "abb_powerone_pvi_sunspec",
  "name": "ABB/Power-One/FIMER PVI SunSpec",
  "codeowners": [
    "@alexdelprete"
  ],
  "config_flow": true,
  "documentation": "https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec",
  "integration_type": "hub",
  "iot_class": "local_polling",
  "issue_tracker": "https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/issues",
  "loggers": [
    "custom_components.abb_powerone_pvi_sunspec"
  ],
  "requirements": [
    "pymodbus>=3.5.4"
  ],
  "version": "v3.5.1-beta.2"
}
```

nonth, 2024-07-31:
@alexdelprete  I'll try to use qmodmaster to check the value at address 40208 and get back to you with the results. 

Also please bear with me as I'm still new to working with Modbus.

nonth, 2024-07-31:
@alexdelprete "I'm currently trying to use qmodmaster. I can connect to my inverter, but I'm not sure how to configure the software to read the value.

Here are the settings I'm using:
- Modbus Mode: TCP
- Unit ID: 247
- Scan Rate (ms): 1000
- Function Code: Read Holding Register (0x03)
- Start Address: 40200 Dec
- Number of Registers: 40
- Data Format: Dec

I'm unsure if the start address and the number of registers are set correctly. from settings above qmodmaster shown "Read data failed. Error Illegal data address"

Can you assist me?

alexdelprete, 2024-08-01:
Assuming you set the correct IP and Unit ID, start with address 40002, read 70 registers, to check if basic info is ok.

[image]

nonth, 2024-08-01:
Yes my inverter ip is 192.168.1.111

Try with address 40002, read 70 registers, still got illegal data address 

[image]

alexdelprete, 2024-08-01:
Do you have a VSN card? If so, the base address is 0. What base address do you use in my integration?

Try address 2 instead of 40002.

alexdelprete, 2024-08-01:
Also, you need to set Big Endian in settings (check the status bar where it says Endian: little in your screenshot).

[image]

nonth, 2024-08-01:
Here's the result after updating the settings:

[image]

alexdelprete, 2024-08-01:
Great, it's working. :)

Can you please start at address 122 reading 80 registers, and post the screenshot?

Thanks.

nonth, 2024-08-01:
address 122 read 80 registers

[image]

but for address 208 got an illegal data address error

[image]

alexdelprete, 2024-08-01:
Can you try from address 200 for 20 registers.

nonth, 2024-08-02:
Here are the result from address 200, read 20 registers. Still no luck :(

[image]

alexdelprete, 2024-08-02:
Your system does not respect the documented register map, apparently.

How many strings do you have?

nonth, 2024-08-02:
I have single inverter connected with 14 solar panels.

alexdelprete, 2024-08-02:
Me too, but I have 27 panels divided on two strings. You have all 14 in 1 string?

nonth, 2024-08-02:
Yes,

Right now, I’m trying to understand how to map the registers from your XLS document with the latest code. I see that offset 1104 is for UNO-DM and REACT2, and offset 208 is for TRIO. What about offset 122?

From my understanding, my TRIO-5.8-TL-OUTD-S inverter with a VSN300 card has been works with offset 122. 

So I’ve tried using 122 as the start address in QModMaster, and I can retrieve data from the inverter (as shown in the screenshot in my previous comment). The first address block has a value of 120, which corresponds to sheet name 120 in SunSpec_PICS-ABB-TRIO-50.0-TL-OUTD.

I’m not sure if I’m on the right track. Any suggestions on what I should do next, or should I try to find the Modbus register mapping for my inverter?

alexdelprete, 2024-08-02:
We don't need Model 120, we need Model 160, which starts at offset 208 (check SunSpec_PICS-ABB-TRIO-50.0-TL-OUTD excel file, the tab 160).

That model contains the values for the strings we need.

alexdelprete, 2025-01-12:
@nonth I haven't received any feedback, so I'll close the issue.

closed 2025-01-12 by alexdelprete