alexdelprete/ha-abb-powerone-pvi-sunspec #217: Unknown error occured

AmelV1, 2025-01-16:
### System Health details

## System Information

version | core-2025.1.2
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.6.66-haos
arch | aarch64
timezone | Europe/Amsterdam
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4988
Installed Version | 2.0.3
Stage | running
Available Repositories | 1500
Downloaded Repositories | 2

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 14.1
-- | --
update_channel | stable
supervisor_version | supervisor-2024.12.3
agent_version | 1.6.0
docker_version | 27.2.0
disk_total | 28.0 GB
disk_used | 4.9 GB
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | green
supervisor_api | ok
version_api | ok
installed_addons | File editor (5.8.0), Get HACS (1.3.1), Mosquitto broker (6.5.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 3
-- | --
resources | 0
views | 1
mode | storage

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 22 September 2024 at 15:55
-- | --
current_recorder_run | 16 January 2025 at 17:34
estimated_db_size | 189.39 MiB
database_engine | sqlite
database_version | 3.47.1

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

Hello. I tried to use this. I don't have a ABB-drive but a Vacon drive. Would this be the cause of the error?

[image]

I was hoping that this would make me able to communicate with the drive. It uses 40000 as register. I like to use the holding registers.

### Reproduction steps

1.
2.
3.
...

### Debug logs

```text
???
```

alexdelprete, 2025-01-16:
I don't understand: you have an ABB/FIMER inverter with a VSN300/700 card? 

I need a debug log to understand what's happening.

AmelV1, 2025-01-16:
Hi Alessandro.
Thank you for taking time to take a look at this.

But I think I have to step out because I think this is all above my level
of knowledge.

I have a Vacon drive and this has an OPT-Ci optionboard. This board should
respond to normal Modbus messages.
I was hoping that your integration would just work even though this is not
an ABB, because I could just fill in te register address.

I don't know how to provide you with a debug-log.

I thought that being a programmer for IEC61131 (structured text) would be
enough for this HA, but it is clearly not.

Thanks again!

In case you are interested here is a link to the board that I use in my
drive..
https://drivecentre.ca/wp-content/uploads/2017/03/Vacon-OPT-CI-manual-NX.pdf

alexdelprete, 2025-01-16:
No worries Amel. If you read the repo documentation, you'll find what is the scope of the integration: it has been developed specifically to get data from ABB/FIMER PVI inverters through the modbus protocol.

It's not a generic integration for modbus, those don't exist.

Hope you find something to integrate your device in HA.

AmelV1, 2025-01-17:
Hi.
Thanks for your time and the reply.
I think I was just too naive and believed that the ABB thing would not
matter and just use the Modbus..
It is a bit frustrating though that there isn't any easy way to use the
modbus in HA.
On my PC for instance, I use Modscan32. Is super easy and works like a
charm.

Thanks again.

ivanfmartinez, 2025-01-17:
@AmelV1 I have some devices that I use with standard modbus integration on homeassistant, If you know what are  the registers  you can use this : https://www.home-assistant.io/integrations/modbus

ivanfmartinez, 2025-01-17:
I used holding register without problem, you must define correctly the types and other parameters. 
Its a standard integragion o HA, is a GENERIC modbus, you have to configure each sensor based on your device.

This kind of thing can be based on some trial and error, but after all mapping is made it will work for long time without troubles. 

If you can read the data in other application try to check all the parameters that are used in the application, sometimes one parameter that you did not put on the yaml is a definition that is default on the other app.  And different terms, some apps with use "endian"  to indicate byte order, this integration uses "swap".

AmelV1, 2025-01-17:
Hello Ivan.
Thanks for reply.
I was trying to send out data from HA to the drive.
Would you be so kind to send me your code?
If not via this channel can send direct to amel431 at gmail

Thanks!