alexdelprete/ha-abb-powerone-pvi-sunspec #226: get two inverters in home assistant

Morvie89, 2025-01-25:
### System Health details

Hello, 
I have two inverters from Abb. React 5.0 and abb trio with a wifi card vsn300. 
I have connected the react with the integration in home assistant. I can't get the trio integrated. 
I always get the error "unknown error occured"
Do u know what i can do ? 

Other questions are
Is the battery status still added to the react?

Greetings

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

.

### Reproduction steps

1.
2.
3.
..

### Debug logs

```text
.
```

alexdelprete, 2025-01-26:
Can you try latest beta and enable debug for the component by putting the component line in logger configuration? Example:

```yaml
logger:
  default: warning
  logs:
    custom_components.abb_powerone_pvi_sunspec: debug
```

Regarding the battery, right now the integration doesn't yet support it.

Morvie89, 2025-01-26:
Hello my React settings

[image]

alexdelprete, 2025-01-27:
> How can I activate debug mode?

If you have the integration already installed and configured, just click "Enable Debug Logging" on the integration page:

[image]

Morvie89, 2025-01-27:
Thank you for your advice:
heres the protocol

2025-01-27 19:55:02.192 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Slave ID: 247
2025-01-27 19:55:02.192 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Base Address: 40000
2025-01-27 19:55:02.193 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Manufacturer: ABB
2025-01-27 19:55:02.193 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Model: -3P81-
2025-01-27 19:55:02.193 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Options: 0x0DED/0xFFCB/
2025-01-27 19:55:02.193 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (opt_notprintable) opt_model: 0x0DED/0xFFCB/ - opt_model_int: 13
2025-01-27 19:55:02.193 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (opt_comm_model) comm_model: REACT2-UNO-5.0-TL
2025-01-27 19:55:02.193 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Version: 2222E
2025-01-27 19:55:02.194 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Sernum: 134032-3P81-4518
2025-01-27 19:55:02.197 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Slave ID: 247
2025-01-27 19:55:02.197 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Base Address: 40000
2025-01-27 19:55:02.197 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Inverter Type (int): 101
2025-01-27 19:55:02.197 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Inverter Type (str): Single Phase
2025-01-27 19:55:02.198 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Total Energy Value Read: 6345080
2025-01-27 19:55:02.198 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Total Energy Previous Value: 6345080
2025-01-27 19:55:02.198 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) DC Current Value read: 0.0
2025-01-27 19:55:02.198 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) DC Voltage Value read: -0.0
2025-01-27 19:55:02.198 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) DC Power Value read: 0
2025-01-27 19:55:02.199 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Temp Oth Value read: 30.3
2025-01-27 19:55:02.199 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Temp Cab Value read: 30.6
2025-01-27 19:55:02.199 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Device Status Value read: No Parameters
2025-01-27 19:55:02.199 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Status Vendor Value read: Wait Sun/Grid
2025-01-27 19:55:02.199 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Completed
2025-01-27 19:55:02.199 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Find M160 for model: REACT2-UNO-5.0-TL at offset: 122
2025-01-27 19:55:02.203 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Model is not 160 - offset: 122 - multi_mppt_id: 120
2025-01-27 19:55:02.203 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Find M160 for model: REACT2-UNO-5.0-TL at offset: 1104
2025-01-27 19:55:02.206 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Model is 160 - offset: 1104 - multi_mppt_id: 160
2025-01-27 19:55:02.206 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Found M160 for model: REACT2-UNO-5.0-TL at offset: 1104
2025-01-27 19:55:02.206 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_160) Model: REACT2-UNO-5.0-TL
2025-01-27 19:55:02.207 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_160) Slave ID: 247
2025-01-27 19:55:02.207 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_160) Base Address: 40000
2025-01-27 19:55:02.207 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_160) Offset: 1104
2025-01-27 19:55:02.211 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_160) mppt_nr 2
2025-01-27 19:55:02.212 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_160) dc1curr: 0.0 Round: 0.0 SF: -1
2025-01-27 19:55:02.212 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_160) dc1volt 3.3
2025-01-27 19:55:02.212 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_160) dc1power 0
2025-01-27 19:55:02.212 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_160) dc2curr: 0.0 Round: 0.0 SF: -1
2025-01-27 19:55:02.212 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_160) dc2volt 0.3
2025-01-27 19:55:02.213 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_160) dc2power 0
2025-01-27 19:55:02.213 DEBUG (SyncWorker_1) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_160) Completed
[... cut ...]
    result = await self._hass.async_add_executor_job(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self.read_sunspec_modbus
        ^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 264, in read_sunspec_modbus
    raise ExceptionError() from exception_error
custom_components.abb_powerone_pvi_sunspec.api.ExceptionError

alexdelprete, 2025-01-27:
The slave_id is not a problem, since it's on a different VSNx00. You can't have two inverters with same slave_id only if they're on the same bus. In your case they're totally separate.

You're using 40000 as base address for Trio, but when using VSN300, as per documentation, you need to set 0 for that config value.

[image]

Morvie89, 2025-01-28:
Hello,
ok io tried with the other parameters, but theres another problem 

2025-01-28 20:17:57.347 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.config_flow] Test connection to 192.168.178.44:502 slave id 2
2025-01-28 20:17:57.347 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.config_flow] Creating API Client
2025-01-28 20:17:57.347 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.config_flow] API Client created: calling get data
2025-01-28 20:17:57.347 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] API Client connect to IP: 192.168.178.44 port: 502 slave id: 2 timeout: 59
2025-01-28 20:17:57.348 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port: opening socket on 192.168.178.44:502 with a 3.0s timeout.
2025-01-28 20:18:00.354 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port (ERROR): port not available on 192.168.178.44:502 - error: 11
2025-01-28 20:18:00.355 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Inverter not ready for Modbus TCP connection

alexdelprete, 2025-01-29:
It means that from the HA host you can't connect to 192.168.178.44 on port 502.

What do you use to run HA? Raspberry? Docker? You need to go to shell inside HA, and telnet IP PORT.

Example: telnet 192.168.178.44 502 and if it connects it means network is ok.

Morvie89, 2025-01-29:
Hallo, 
ich habe einen INTEL NUK und lasse HAOS über Linux laufen.

Ich habe immer verssucht die Verbindung aufzubauen, als es Tag war und der Wechselrichter über die IP Adresse erreichbar ist.

Können Sie mir das genauer erklären ?
Sie müssen die Shell innerhalb von HA aufrufen und den IP PORT telnet.

Beispiel: telnet 192.168.178.44 502 und wenn es eine Verbindung herstellt, bedeutet das, dass das Netzwerk in Ordnung ist.

Dafür bin ich noch zu neu in Home Assistant 

Danke schon für die vielen Hilfen :)

alexdelprete, 2025-01-29:
You need to ensure there's network connectivity between the HA host and the VSN300.

Install SSH addon in HAOS, that will give you access to the terminal shell. From there, ping the vsn300 and telnet on port 502.

Morvie89, 2025-01-29:
I dont really know what u mean with telnet. 

i tryied following:

[image]

alexdelprete, 2025-01-29:
The integration connects on an IP and a port. You need to check if you have connectivity on that specific port.

write this command instead of ping:

telnet 192.168.178.44 502

and give me the output of the command.

I don't know how to explain it simpler than this. :)

Morvie89, 2025-01-30:
Hello,
what im doing Wrong ?

[image]

alexdelprete, 2025-01-30:
Ok, try this other command: `nc -zv 192.168.178.44 502`

Morvie89, 2025-01-31:
Hello, i dont know wheather i do something wrong ? 

[image]

alexdelprete, 2025-01-31:
you put spaces between command and option. you wrote `nc - z`, copy paste the command as I gave it to you in the post: `nc -zv 192.168.178.44 502`

There is a space after nc, a space after zv and a space after 44. :)

Morvie89, 2025-02-01:
Ok ot says open :) 

[image]

Morvie89, 2025-02-01:
tried again with degug log:

There's one Problem:
2025-02-01 12:45:21.651 DEBUG (SyncWorker_3) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Received Modbus library exception: Exception Response(131, 3, IllegalAddress)

But i dont know what meant by illegal Address ? 

custom_components.abb_powerone_pvi_sunspec.api.ConnectionError: Inverter not active on 192.168.178.44:502
2025-02-01 12:45:19.236 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Data Coordinator: Update started at 2025-02-01 12:45:19.236426
2025-02-01 12:45:19.236 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] API Client connect to IP: 192.168.178.45 port: 502 slave id: 247 timeout: 29
2025-02-01 12:45:19.237 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port: opening socket on 192.168.178.45:502 with a 3.0s timeout.
2025-02-01 12:45:21.375 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port (SUCCESS): port open on 192.168.178.45:502
2025-02-01 12:45:21.376 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Inverter ready for Modbus TCP connection
2025-02-01 12:45:21.383 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Modbus TCP Client connected
2025-02-01 12:45:21.383 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Start Get data (Slave ID: 247 - Base Address: 40000)
2025-02-01 12:45:21.398 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Slave ID: 247
2025-02-01 12:45:21.398 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Base Address: 40000
2025-02-01 12:45:21.398 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Manufacturer: ABB
2025-02-01 12:45:21.398 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Model: -3P81-
2025-02-01 12:45:21.398 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Options: 0x0DED/0xFFCB/
2025-02-01 12:45:21.398 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (opt_notprintable) opt_model: 0x0DED/0xFFCB/ - opt_model_int: 13
2025-02-01 12:45:21.398 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (opt_comm_model) comm_model: REACT2-UNO-5.0-TL
2025-02-01 12:45:21.398 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Version: 2222E
2025-02-01 12:45:21.398 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Sernum: 134032-3P81-4518
2025-02-01 12:45:21.405 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.config_flow] Test connection to 192.168.178.44:502 slave id 2
2025-02-01 12:45:21.405 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.config_flow] Creating API Client
2025-02-01 12:45:21.405 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.config_flow] API Client created: calling get data
2025-02-01 12:45:21.405 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] API Client connect to IP: 192.168.178.44 port: 502 slave id: 2 timeout: 59
2025-02-01 12:45:21.405 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port: opening socket on 192.168.178.44:502 with a 3.0s timeout.
2025-02-01 12:45:21.408 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port (SUCCESS): port open on 192.168.178.44:502
2025-02-01 12:45:21.408 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Inverter ready for Modbus TCP connection
2025-02-01 12:45:21.411 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Slave ID: 247
2025-02-01 12:45:21.411 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Base Address: 40000
2025-02-01 12:45:21.411 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Inverter Type (int): 101
2025-02-01 12:45:21.412 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Inverter Type (str): Single Phase
2025-02-01 12:45:21.412 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Total Energy Value Read: 6374320
2025-02-01 12:45:21.412 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Total Energy Previous Value: 6374290
2025-02-01 12:45:21.412 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) DC Current Value read: 10.4
2025-02-01 12:45:21.412 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) DC Voltage Value read: -0.0
2025-02-01 12:45:21.412 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) DC Power Value read: 2510
2025-02-01 12:45:21.412 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Temp Oth Value read: 41.3
2025-02-01 12:45:21.412 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Temp Cab Value read: 33.7
2025-02-01 12:45:21.412 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Device Status Value read: Out OC
2025-02-01 12:45:21.412 DEBUG (SyncWorker_9) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Status Vendor Value read: Run
[... cut ...]
2025-02-01 12:45:21.651 DEBUG (SyncWorker_3) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Received Modbus library exception: Exception Response(131, 3, IllegalAddress)
2025-02-01 12:45:21.651 DEBUG (SyncWorker_3) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Generic error: 
2025-02-01 12:45:21.651 DEBUG (SyncWorker_3) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Generic error: 
2025-02-01 12:45:21.653 ERROR (MainThread) [aiohttp.server] Error handling request

alexdelprete, 2025-02-02:
So you are able to install the integration? You said you weren't able to install it.

I don't understand these logs, I think you're mixing logs from different tests. Also make sure you're using latest integration version.

Please test with only 1 inverter at a time. And provide the logs only for that test.

Morvie89, 2025-02-13:
Hello,
I think we've been talking at cross purposes here.
Basically, I am trying to integrate two inverters into the integration.
React 5 and TRIO

I can integrate React 5. This is stored under the IP 192.168.178.44:502

the TRIO is the problem. This is under the IP 192.168.178.45 

Is it even possible to integrate both inverters?

alexdelprete, 2025-02-13:
Test one inverter at a time, like I already told you. If REACT works, delete everything and configure ONLY the TRIO, and show me the debug log.

From the logs above it seems that REACT is .45, but it's a confused log. Add only the TRIO and let's see the debug log.

In principle, the integration supports multiple inverters, but I can't test it since I only have one.

Morvie89, 2025-02-13:
I hope that helps:

2025-02-13 14:38:15.338 ERROR (SyncWorker_3) [pymodbus.logging] Exception response 131 / 0
2025-02-13 14:38:15.340 ERROR (MainThread) [aiohttp.server] Error handling request from 192.168.178.38
Traceback (most recent call last):
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 355, in read_sunspec_modbus_model_1
    raise ModbusError()
custom_components.abb_powerone_pvi_sunspec.api.ModbusError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 250, in read_sunspec_modbus
    self.read_sunspec_modbus_model_1()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 366, in read_sunspec_modbus_model_1
    raise ExceptionError() from exception_error
custom_components.abb_powerone_pvi_sunspec.api.ExceptionError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 222, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 121, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 340, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 387, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... 84 lines cut ...]
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        flow, cur_step["step_id"], user_input
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 490, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/abb_powerone_pvi_sunspec/config_flow.py", line 138, in async_step_user
    uid = await self.get_unique_id(
          ^^^^^^^^^^^^^^^^^^^^^^^^^
        name, host, port, slave_id, base_addr, scan_interval
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/abb_powerone_pvi_sunspec/config_flow.py", line 111, in get_unique_id
    self.api_data = await self.api.async_get_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 220, in async_get_data
    if self.connect():
       ~~~~~~~~~~~~^^
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 195, in connect
    raise ConnectionError(f"Inverter not active on {self._host}:{self._port}")
custom_components.abb_powerone_pvi_sunspec.api.ConnectionError: Inverter not active on 192.168.178.100:502

alexdelprete, 2025-02-13:
You told me the inverters are on .44 and .45, in last log I see 192.168.178.38. Wrong IP.

Please make sure to do things properly, write the correct ip when configuring, and also the other parameters (slave_id, base address, etc.)

Morvie89, 2025-02-13:
192.168.178.38 is my PC at Network
i saw the Invertrer changed the IP to 192.168.178.100. 
I gave now a Static IP 192.168.178.100.  and dont get it :/

Morvie89, 2025-02-13:
2025-02-13 15:04:05.770 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.config_flow] Test connection to 192.168.178.100:502 slave id 2
2025-02-13 15:04:05.770 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.config_flow] Creating API Client
2025-02-13 15:04:05.771 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.config_flow] API Client created: calling get data
2025-02-13 15:04:05.771 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] API Client connect to IP: 192.168.178.100 port: 502 slave id: 2 timeout: 59
2025-02-13 15:04:05.771 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port: opening socket on 192.168.178.100:502 with a 3.0s timeout.
2025-02-13 15:04:05.775 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port (SUCCESS): port open on 192.168.178.100:502
2025-02-13 15:04:05.776 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Inverter ready for Modbus TCP connection
2025-02-13 15:04:05.976 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Modbus TCP Client connected
2025-02-13 15:04:05.976 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Start Get data (Slave ID: 2 - Base Address: 0)
2025-02-13 15:04:06.018 ERROR (SyncWorker_4) [pymodbus.logging] Exception response 131 / 0
2025-02-13 15:04:06.018 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Received Modbus library exception: ExceptionResponse(dev_id=2, transaction_id=1, address=0, count=0, bits=[], registers=[], status=1)
2025-02-13 15:04:06.019 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Generic error: 
2025-02-13 15:04:06.019 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_sunspec_modbus) Generic error: 
2025-02-13 15:04:06.022 ERROR (MainThread) [aiohttp.server] Error handling request from 192.168.178.38
Traceback (most recent call last):
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 355, in read_sunspec_modbus_model_1
    raise ModbusError()
custom_components.abb_powerone_pvi_sunspec.api.ModbusError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 250, in read_sunspec_modbus
    self.read_sunspec_modbus_model_1()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 366, in read_sunspec_modbus_model_1
    raise ExceptionError() from exception_error
custom_components.abb_powerone_pvi_sunspec.api.ExceptionError

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 480, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 32, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 81, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 222, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
[... 14 lines cut ...]
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 490, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... cut ...]
    result = await self._hass.async_add_executor_job(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self.read_sunspec_modbus
        ^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 272, in read_sunspec_modbus
    raise ExceptionError() from exception_error
custom_components.abb_powerone_pvi_sunspec.api.ExceptionError

alexdelprete, 2025-02-13:
Ok, the ip .100 seems correct. Probably the slave_id (you used 2) or the base address (you used 0) are wrong. Are you using the VSN300 on the TRIO?

Morvie89, 2025-02-13:
Yes

alexdelprete, 2025-02-14:
If you use the VSN300, base address should be 0. But you need to find the inverter's slave_id. You can check for it in the inverter LCD display. 

Also: did you enable modbus tcp on the VSN300? In the web configuration of the VSN300.

Morvie89, 2025-02-14:
Oh wow,
I see the problem. I opened the inverter and saw that it is not the VSN 300, but the Control Unit Ethernet Expansion Board.
That probably means I can't integrate it.

ivanfmartinez, 2025-02-14:
For reference I use two inverters with this integration for a long time.

alexdelprete, 2025-02-14:
> Control Unit Ethernet Expansion Board

you can install the VSN300.

alexdelprete, 2025-02-14:
> For reference I use two inverters with this integration for a long time.

Thank you Ivan.

Morvie89, 2025-02-17:
Hello,
I am so grateful to get so much help here.
Sorry again for trying with the wrong card. (Ethernet Board)

I have now bought and installed a VSN 300
It's so hard that it still doesn't work.
I get to the interface and have activated the TCP Modbus. Slave ID is set to 2:
However, I still cannot integrate the inverter in Home Assistant:
I have deleted the React and iam starting to set up the Trio again:
However, I get the following feedback:

2025-02-17 16:12:21.096 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.config_flow] Test connection to 192.168.178.77:502 slave id 2
2025-02-17 16:12:21.096 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.config_flow] Creating API Client
2025-02-17 16:12:21.096 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.config_flow] API Client created: calling get data
2025-02-17 16:12:21.097 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] API Client connect to IP: 192.168.178.77 port: 502 slave id: 2 timeout: 29
2025-02-17 16:12:21.097 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port: opening socket on 192.168.178.77:502 with a 3.0s timeout.
2025-02-17 16:12:21.385 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port (SUCCESS): port open on 192.168.178.77:502
2025-02-17 16:12:21.386 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Inverter ready for Modbus TCP connection
2025-02-17 16:12:21.390 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Modbus TCP Client connected
2025-02-17 16:12:21.390 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Start Get data (Slave ID: 2 - Base Address: 0)
2025-02-17 16:14:17.485 ERROR (SyncWorker_7) [pymodbus.logging] No response received after 3 retries, continue with next request
2025-02-17 16:14:17.486 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] Read Holding Registers modbus_error: Modbus Error: [Input/Output] No response received after 3 retries, continue with next request
2025-02-17 16:14:17.486 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Generic error: 
2025-02-17 16:14:17.486 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (read_sunspec_modbus) Generic error

Why is the inverter not responding ?

Best Regards

Morvie89, 2025-02-17:
[image]

[image]

Morvie89, 2025-02-17:
[image]

alexdelprete, 2025-02-17:
Check that the inverter is actually 2 as slave id. I don't know if that `RS485 Address line 1` is correct. Check on the inverter, on the display.

alexdelprete, 2025-02-17:
You can also see it in the cloud: www.auroravision.net.

[image]

Morvie89, 2025-02-17:
oh yeah 
u are a Genie!
it was Adress 243. I dont know why. In my Inverter it says another Slave ID.
I got it now
Thank you :=)

Is der a plan to get the Batterie from React to Home Assistant too ?

alexdelprete, 2025-02-17:
Please close the issue if you're ok now. :)

Morvie89, 2025-02-17:
Im Really thank you :=)

alexdelprete, 2025-02-25:
@Morvie89 can you please post a debug log for the TRIO? I need to check some things for other TRIO users. Thanks.