alexdelprete/ha-abb-powerone-pvi-sunspec #240: DC1, DC2 sensors are unavailable from 3-phase Inverter

nonth, 2025-02-24:
### System Health details

## System Information

version | core-2025.2.4
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.13.1
os_name | Linux
os_version | 6.8.0-52-generic
arch | x86_64
timezone | Asia/Bangkok
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4988
Installed Version | 2.0.5
Stage | running
Available Repositories | 1553
Downloaded Repositories | 5

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | January 1, 2018 at 7:00 AM
relayer_connected | false
relayer_region | null
remote_enabled | true
remote_connected | false
alexa_enabled | false
google_enabled | false
cloud_ice_servers_enabled | true
remote_server | null
certificate_status | null
instance_id | 39fe8ae6d4a44c54a5102898879b3151
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 11
-- | --
resources | 9
views | 22
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), eth0 (enabled, default, auto)
-- | --
ipv4_addresses | lo (127.0.0.1/8), eth0 (172.21.0.6/24)
[... 128 lines cut ...]
2025-02-24 10:54:34.010 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Base Address: 0
2025-02-24 10:54:34.010 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Inverter Type (int): 103
2025-02-24 10:54:34.010 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Inverter Type (str): Three Phase
2025-02-24 10:54:34.010 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Total Energy Value Read: 29288384
2025-02-24 10:54:34.010 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Total Energy Previous Value: 29288356
2025-02-24 10:54:34.010 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) DC Power Value read: 1384
2025-02-24 10:54:34.010 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Temp Oth Value read: 41.6
2025-02-24 10:54:34.010 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Temp Cab Value read: 39.3
2025-02-24 10:54:34.010 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Device Status Value read: Out OC
2025-02-24 10:54:34.010 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Status Vendor Value read: Run
2025-02-24 10:54:34.011 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_101_103) Completed
2025-02-24 10:54:34.011 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Find M160 for model: TRIO-5.8-OUTD at offset: 122
2025-02-24 10:54:34.017 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Model is not 160 - offset: 122 - multi_mppt_id: 120
2025-02-24 10:54:34.017 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Find M160 for model: TRIO-5.8-OUTD at offset: 1104
2025-02-24 10:54:34.023 ERROR (SyncWorker_7) [pymodbus.logging] Exception response 131 / 0
2025-02-24 10:54:34.023 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Received Modbus library exception: ExceptionResponse(dev_id=247, transaction_id=9, address=0, count=0, bits=[], registers=[], status=1)
2025-02-24 10:54:34.023 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Model is not 160 - offset: 1104 - multi_mppt_id: 120
2025-02-24 10:54:34.023 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Find M160 for model: TRIO-5.8-OUTD at offset: 208
2025-02-24 10:54:34.030 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Model is not 160 - offset: 208 - multi_mppt_id: 65535
2025-02-24 10:54:34.030 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) M160 not found for model: TRIO-5.8-OUTD
2025-02-24 10:54:34.030 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] read_sunspec_modbus: success True
2025-02-24 10:54:34.032 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Closing Modbus TCP connection
2025-02-24 10:54:34.032 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] End Get data
2025-02-24 10:54:34.032 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Get Data Result: valid
```

alexdelprete, 2025-02-25:
Were those 4 sensors available in previous release? If so, can you please rollback to previous working version and provide a debug log?

The issue here is that it doesn't find the starting offset for Sunspec Model 160 data at none of the known offsets:

```python
2025-02-24 10:53:33.565 DEBUG (SyncWorker_2) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Find M160 for model: TRIO-5.8-OUTD at offset: 122
2025-02-24 10:53:33.579 DEBUG (SyncWorker_2) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Model is not 160 - offset: 122 - multi_mppt_id: 120
2025-02-24 10:53:33.579 DEBUG (SyncWorker_2) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Find M160 for model: TRIO-5.8-OUTD at offset: 1104
2025-02-24 10:53:33.591 ERROR (SyncWorker_2) [pymodbus.logging] Exception response 131 / 0
2025-02-24 10:53:33.591 DEBUG (SyncWorker_2) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Received Modbus library exception: ExceptionResponse(dev_id=247, transaction_id=4, address=0, count=0, bits=[], registers=[], status=1)
2025-02-24 10:53:33.591 DEBUG (SyncWorker_2) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Model is not 160 - offset: 1104 - multi_mppt_id: 120
2025-02-24 10:53:33.591 DEBUG (SyncWorker_2) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Find M160 for model: TRIO-5.8-OUTD at offset: 208
2025-02-24 10:53:33.622 DEBUG (SyncWorker_2) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) Model is not 160 - offset: 208 - multi_mppt_id: 65535
2025-02-24 10:53:33.622 DEBUG (SyncWorker_2) [custom_components.abb_powerone_pvi_sunspec.api] (find_m160) M160 not found for model: TRIO-5.8-OUTD
```

alexdelprete, 2025-02-25:
I remember having another discussion with you in #165. We weren't able to find the offset for M160 in January and had no feedback from you after last post. So how did you have those 4 sensors available without being able to read M160 data?

nonth, 2025-02-25:
Yes we have a discussion before like you mention. the version that work fine without any issue for me is version before v3.3.0 as I mentioned in #165 

at that time I decided to stick with v3.1.5 until I found that you have refactor M160 model offset detection (v3.5.5) so I give it a try on a new version. this time integration can read data from inverter except those 4 sensors 

Do you need any information from my side that might help to solve this issue ?

alexdelprete, 2025-02-25:
> Yes we have a discussion before like you mention. the version that work fine without any issue for me is version before v3.3.0 as I mentioned in [#165](https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/issues/165)

I doubt that with that version you had those 4 sensors working, since M160 is not available at known offsets.

Anyway, you can try rolling back to the version that worked for you and capture a debug log, so we can see what it's doing to get M160 data.

nonth, 2025-02-26:
I tried rolling back to a version v3.1.5 that was working, but I ran into this error:  

`ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 were given (plus 1 keyword-only argument).`  

Not sure if this is related to the latest Home Assistant update (just upgraded to v2025.2.4). I'll try to get it working and send you the logs.

nonth, 2025-02-26:
Here is a log from v3.5.0, in which I've modified the read_holding_registers function to make it work with pymodbus>=3.7.4. (with some warning)

```
def read_holding_registers(self, slave, address, count):
        """Read holding registers."""
        kwargs = {"slave": slave} if slave else {}
        try:
            with self._lock:
                # return self._client.read_holding_registers(address, count, **kwargs)

                # modify to make it work with pymodbus>=3.7.4
                return self._client.read_holding_registers(
                    address=address, count=count, slave=slave
                )
        except ConnectionException as connect_error:
            _LOGGER.debug(f"Read Holding Registers connect_error: {connect_error}")
            raise ConnectionError() from connect_error
        except ModbusException as modbus_error:
            _LOGGER.debug(f"Read Holding Registers modbus_error: {modbus_error}")
            raise ModbusError() from modbus_error
```

```
2025-02-26 10:08:59.980 INFO (MainThread) [custom_components.abb_powerone_pvi_sunspec] 
-------------------------------------------------------------------
ABB/Power-One/FIMER PVI SunSpec ModBus TCP
Version: 3.0.0
by @alexdelprete
This is a custom integration!
If you have any issues with this you need to open an issue here:
https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/issues
-------------------------------------------------------------------

2025-02-26 10:08:59.980 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Setup config_entry for abb_powerone_pvi_sunspec
2025-02-26 10:08:59.980 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Scan Interval: scan_interval=60.0 update_interval=0:01:00
2025-02-26 10:08:59.981 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Coordinator Config Data: {'base_addr': 0, 'host': '192.168.1.111', 'name': 'VSN300', 'port': 502, 'scan_interval': 60.0, 'slave_id': 247}
2025-02-26 10:08:59.981 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Coordinator API init: Host: 192.168.1.111 Port: 502 ID: 247 ScanInterval: 60.0
2025-02-26 10:08:59.981 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Data Coordinator: Update started at 2025-02-26 10:08:59.981186
2025-02-26 10:08:59.981 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] API Client connect to IP: 192.168.1.111 port: 502 slave id: 247 timeout: 59.0
2025-02-26 10:08:59.981 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port: opening socket on 192.168.1.111:502 with a 3.0s timeout.
2025-02-26 10:09:00.052 WARNING (ImportExecutor_0) [py.warnings] /usr/local/lib/python3.13/site-packages/miio/miot_device.py:23: FutureWarning: functools.partial will be a method descriptor in future Python versions; wrap it in enum.member() if you want to preserve the old behavior
  Bool = partial(_str2bool)

2025-02-26 10:09:00.246 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port (SUCCESS): port open on 192.168.1.111:502
2025-02-26 10:09:00.246 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Inverter ready for Modbus TCP connection
2025-02-26 10:09:00.261 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Modbus TCP Client connected
2025-02-26 10:09:00.261 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Start Get data (Slave ID: 247 - Base Address: 0)
2025-02-26 10:09:00.320 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Slave ID: 247
2025-02-26 10:09:00.321 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Base Address: 0
2025-02-26 10:09:00.321 WARNING (SyncWorker_4) [pymodbus.logging] BinaryPayloadDecoder is deprecated and will be removed in v3.9.0 !
Please use "client.convert_from_registers()" or "client.convert_to_registers"
See documentation: "https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.mixin.ModbusClientMixin.convert_from_registers"
2025-02-26 10:09:00.322 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Manufacturer: Power-One
2025-02-26 10:09:00.322 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Model: -3M97-
2025-02-26 10:09:00.322 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Options: r
2025-02-26 10:09:00.322 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (opt_printable) opt_model: r - opt_model_int: 114
2025-02-26 10:09:00.322 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (opt_comm_model) comm_model: TRIO-5.8-OUTD
2025-02-26 10:09:00.322 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Version: CAA8
2025-02-26 10:09:00.322 DEBUG (SyncWorker_4) [custom_components.abb_powerone_pvi_sunspec.api] (read_rt_1) Sernum: 108624-3M97-5019
2025-02-26 10:09:00.338 WARNING (ImportExecutor_0) [homeassistant.components.dhcp] DhcpServiceInfo was used from tapo, this is a deprecated constant which will be removed in HA Core 2026.2. Use homeassistant.helpers.service_info.dhcp.DhcpServiceInfo instead, please report it to the author of the 'tapo' custom integration
[... cut ...]
2025-02-26 10:12:02.027 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Data Coordinator: Update completed at 2025-02-26 10:12:02.027436
2025-02-26 10:12:02.027 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Finished fetching abb_powerone_pvi_sunspec (108624-3M97-5019) data in 0.054 seconds (success: True)
2025-02-26 10:12:02.027 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.sensor] _handle_coordinator_update: sensors state written to state machine
```

nonth, 2025-02-26:
I've noticed that before upgrading to v3.6.0, the values of DC1 and DC2 current, power, and voltage were always 1A, 1W, and 1V. I'm not sure if this is because it cannot find the offset for M160, but in the previous version, it would still show 1A, 1W, and 1V. However, in v3.6.0, when it cannot find the offset, it shows 'unavailable.' Is there any way I can find the correct offset?

[image]

alexdelprete, 2025-02-26:
The offset for M160 was not found also with old versions. The behaviour of latest integration version is correct: those 4 sensors are not available because M160 data is not retrieved, so they are orphaned now. You can delete them.

In order to find the offset, you can use qmodmaster and test all the memory pages until you find value 160.

Can't do much for you, this is the same discussion we had in #165.

closed 2025-04-23 by alexdelprete