alexdelprete/ha-abb-powerone-pvi-sunspec #252: After updating HA to 2025.4.1 'read_sunspec_modbus' fails with an error due to ModbusClientMixin.read_holding_registers() missing arguments

goberhammer, 2025-04-12:
### System Health details

## System Information

version | core-2025.4.1
-- | --
installation_type | Home Assistant Container
dev | false
hassio | false
docker | true
user | root
virtualenv | false
python_version | 3.13.2
os_name | Linux
os_version | 5.15.0-136-generic
arch | x86_64
timezone | Europe/Rome
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4994
Installed Version | 2.0.5
Stage | running
Available Repositories | 1647
Downloaded Repositories | 1

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Dashboards</summary>

dashboards | 4
-- | --
resources | 4
views | 20
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp6s18 (enabled, default, auto), docker0 (disabled), br-9488e6b44172 (disabled), vethaeb843e (disabled), veth0ac77e6 (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp6s18 (192.168.60.105/24), docker0 (172.17.0.1/16), br-9488e6b44172 (172.18.0.1/16), vethaeb843e (), veth0ac77e6 ()
ipv6_addresses | lo (::1/128), enp6s18 (fe80::b49e:1dff:fe11:2642/64), docker0 (), br-9488e6b44172 (fe80::42:14ff:fefb:40eb/64), vethaeb843e (fe80::94a1:eeff:fedd:9696/64), veth0ac77e6 (fe80::74cb:33ff:feff:2859/64)
announce_addresses | 192.168.60.105, fe80::b49e:1dff:fe11:2642

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 25 marzo 2025 alle ore 15:16
-- | --
current_recorder_run | 10 aprile 2025 alle ore 15:43
estimated_db_size | 406.08 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

After updating to HA 2025.4.1 integration can't fetch information from inverter anymore.
Debug log shows that the modbus TCP connection is establiched correctly but subsequent calls to 'read_sunspec_modbus' fail with error:
`ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given`

I suspect some internal API changed.

### Reproduction steps

1. Install integration
2. Configure a TCP modbus connecttion to an ABB inverter
3. Restart HA

### Debug logs

```text
2025-04-12 09:25:40.228 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Setup config_entry for abb_powerone_pvi_sunspec
2025-04-12 09:25:40.228 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Scan Interval: scan_interval=60 update_interval=0:01:00
2025-04-12 09:25:40.228 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Coordinator Config Data: {'base_addr': 40000, 'host': '192.168.80.14', 'name': 'FimerUnoDm3Plus', 'port': 502, 'scan_interval': 60, 'slave_id': 1}
2025-04-12 09:25:40.228 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Coordinator API init: Host: 192.168.80.14 Port: 502 ID: 1 ScanInterval: 60
2025-04-12 09:25:40.228 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Data Coordinator: Update started at 2025-04-12 09:25:40.228354
2025-04-12 09:25:40.228 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] API Client connect to IP: 192.168.80.14 port: 502 slave id: 1 timeout: 59
2025-04-12 09:25:40.228 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port: opening socket on 192.168.80.14:502 with a 3.0s timeout.
2025-04-12 09:25:40.264 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port (SUCCESS): port open on 192.168.80.14:502
2025-04-12 09:25:40.265 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Inverter ready for Modbus TCP connection
2025-04-12 09:25:40.277 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Modbus TCP Client connected
2025-04-12 09:25:40.277 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Start Get data (Slave ID: 1 - Base Address: 40000)
2025-04-12 09:25:40.277 DEBUG (SyncWorker_7) [custom_components.abb_powerone_pvi_sunspec.api] read_sunspec_modbus: failed with error: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
2025-04-12 09:25:40.278 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Coordinator Update Error:  at 2025-04-12 09:25:40.228182
2025-04-12 09:25:40.278 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Finished fetching abb_powerone_pvi_sunspec (104174-3P86-0721) data in 0.050 seconds (success: False)
2025-04-12 09:25:45.440 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Setup config_entry for abb_powerone_pvi_sunspec
2025-04-12 09:25:45.440 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Scan Interval: scan_interval=60 update_interval=0:01:00
2025-04-12 09:25:45.440 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Coordinator Config Data: {'base_addr': 40000, 'host': '192.168.80.14', 'name': 'FimerUnoDm3Plus', 'port': 502, 'scan_interval': 60, 'slave_id': 1}
2025-04-12 09:25:45.440 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Coordinator API init: Host: 192.168.80.14 Port: 502 ID: 1 ScanInterval: 60
2025-04-12 09:25:45.440 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Data Coordinator: Update started at 2025-04-12 09:25:45.440689
2025-04-12 09:25:45.440 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] API Client connect to IP: 192.168.80.14 port: 502 slave id: 1 timeout: 59
2025-04-12 09:25:45.440 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port: opening socket on 192.168.80.14:502 with a 3.0s timeout.
2025-04-12 09:25:45.445 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Check_Port (SUCCESS): port open on 192.168.80.14:502
2025-04-12 09:25:45.445 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Inverter ready for Modbus TCP connection
2025-04-12 09:25:45.451 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Modbus TCP Client connected
2025-04-12 09:25:45.451 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Start Get data (Slave ID: 1 - Base Address: 40000)
2025-04-12 09:25:45.451 DEBUG (SyncWorker_0) [custom_components.abb_powerone_pvi_sunspec.api] read_sunspec_modbus: failed with error: ModbusClientMixin.read_holding_registers() takes 2 positional arguments but 3 positional arguments (and 1 keyword-only argument) were given
2025-04-12 09:25:45.452 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Coordinator Update Error:  at 2025-04-12 09:25:45.440533
2025-04-12 09:25:45.452 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Finished fetching abb_powerone_pvi_sunspec (104174-3P86-0721) data in 0.011 seconds (success: False)
```

alexdelprete, 2025-04-13:
I'm on `2025.4.2`, and everything's working fine. HA `2025.4.1` and `2025.4.2` use `pymodbus 3.8.3`.

This is the description of `read_holding_registers()` from [pymodbus 3.8.3 docs](https://pymodbus.readthedocs.io/en/v3.8.3/source/client.html#pymodbus.client.mixin.ModbusClientMixin.read_holding_registers):

[image]

And this is the function definition in my code:

```python
    def read_holding_registers(self, address, count):
        """Read holding registers."""

        try:
            with self._lock:
                return self._client.read_holding_registers(
                    address=address, count=count, slave=self._slave_id
                )
        except ConnectionException as connect_error:
            _LOGGER.debug(f"Read Holding Registers connect_error: {connect_error}")
            raise ConnectionError() from connect_error
        except ModbusException as modbus_error:
            _LOGGER.debug(f"Read Holding Registers modbus_error: {modbus_error}")
            raise ModbusError() from modbus_error
```

As you can see, I'm using keyword parameters, exactly as described in docs.

I suspect you're using some other integrations which is hard-pinning a higher version of pymodbus, that is not in line with HA requirements (pymodbus==3.8.3).

goberhammer, 2025-04-13:
Thanks for the quick and useful response.

Thanks to the snipped of code you posted above I discovered that I was running a different (older) version of your integration.

The strange thing is that HACS was not suggesting to update it (as it usually does) so I missed an update.

Forced install of latest version and now everything is fine.

Sorry for the noise and thanks again :-)

closed 2025-04-13 by goberhammer

alexdelprete, 2025-04-13:
I'm sorry, should've asked what version of the integration you were using, I gave for granted you were on the latest. :)

alexdelprete, 2025-04-14:
Non ci si annoia mai con HA, è peggio del mantenere una casa: manutenzione ogni settimana. :)