alexdelprete/ha-abb-powerone-pvi-sunspec #287: Abient temperature over 50 degrees are scaled wrong

kairisku, 2025-07-28:
### System Health details

-

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

TRIO-8.5-TL-OUTD-S inverter with VSN300 datalogger card. Then ambient temperature goes above 50 degrees, it is wrongly scaled to one tenth of the actual value. Inverter temperature is seem correct the whole time. See example image below:

[image]

### Reproduction steps

1. Wait for a hot day.
2. Observe the readings.

### Debug logs

```text
-
```

### Diagnostics dump

-

alexdelprete, 2025-08-11:
On some inverters, the scale factor reported by the inverter for cabinet temperature is wrong, so I introduced a high limit after which I reduced the scale factor.

I thought 50° was a good threshold, but I didn't think about really hot weather conditions in some countries. :)

I raised it to 70° now. If you want to test it immediately, before I release a new version, modify line 66 in const.py like this:

```python
# Temperature processing constants
TEMP_THRESHOLD_CELSIUS = 70
```

kairisku, 2025-08-11:
Finland is not exactly known for very high temperatures, but we had a few extreme weeks this summer that happened just as I had started using this integration. I have not seen the threshold exceeded now for a couple of weeks, so it might not happen until next summer. But I modified the code as instructed. 

Thank you for your attention in this matter!

alexdelprete, 2025-08-11:
> Panels are on a small hill with the inverter right behind them, so already a warm place in summer. There is no external sensor for ambient temperature so it must be somewhere at the bottom of the inverter closure and affected by inverter heat as well.

That makes sense. What about winter season? Wouldn't it be too cold for that type of installation?