alexdelprete/ha-abb-powerone-pvi-sunspec #296: With Home Assistant 2025.9.0 the integration fails to load

Jump-333, 2025-09-04:
### System Health details

## System Information

version | core-2025.9.0
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
container_arch | amd64
user | root
virtualenv | false
python_version | 3.13.7
os_name | Linux
os_version | 6.12.41-haos
arch | x86_64
timezone | Europe/Rome
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4998
Installed Version | 2.0.5
Stage | running
Available Repositories | 2167
Downloaded Repositories | 16

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | false
-- | --
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 16.1
-- | --
update_channel | stable
supervisor_version | supervisor-2025.08.3
agent_version | 1.7.2
docker_version | 28.3.3
disk_total | 219.4 GB
disk_used | 18.7 GB
nameservers | 10.0.2.1, fd35:b763:f212:0:62b5:8dff:fe4e:a9
healthy | true
supported | true
host_connectivity | true
supervisor_connectivity | true
ntp_synchronized | true
virtualization | 
board | generic-x86-64
supervisor_api | ok
version_api | ok
installed_addons | Matter Server (8.1.0), File editor (5.8.0), Advanced SSH & Web Terminal (21.0.2), Samba share (12.5.2), Samba Backup (5.2.0)

</details>

<details><summary>Dashboards</summary>

dashboards | 10
-- | --
resources | 11
views | 39
mode | storage

</details>

<details><summary>Network Configuration</summary>

adapters | lo (disabled), enp1s0f0 (enabled, default, auto), hassio (disabled), docker0 (disabled), veth2263ce3 (disabled), vethee95ff8 (disabled), veth22a030d (disabled), veth1ec1753 (disabled), vethd610d42 (disabled), vethd8ff0e5 (disabled), vethb7ad39f (disabled)
-- | --
ipv4_addresses | lo (127.0.0.1/8), enp1s0f0 (10.0.2.131/24), hassio (172.30.32.1/23), docker0 (172.30.232.1/23), veth2263ce3 (), vethee95ff8 (), veth22a030d (), veth1ec1753 (), vethd610d42 (), vethd8ff0e5 (), vethb7ad39f ()
ipv6_addresses | lo (::1/128), enp1s0f0 (fd35:b763:f212:0:634:78a9:db30:12cf/64, fe80::2004:c8f4:b08b:3c70/64), hassio (fe80::dc1b:f2ff:fea4:e210/64), docker0 (fe80::4862:cdff:fede:4171/64), veth2263ce3 (fe80::10e1:8ff:fed7:7c16/64), vethee95ff8 (fe80::14a1:2ff:fea4:2b39/64), veth22a030d (fe80::f819:33ff:fe28:284c/64), veth1ec1753 (fe80::1c4d:3dff:feef:5416/64), vethd610d42 (fe80::94da:f3ff:feee:ecda/64), vethd8ff0e5 (fe80::e828:1dff:fe7f:7f6a/64), vethb7ad39f (fe80::d8c1:f3ff:fec3:97fb/64)
announce_addresses | 10.0.2.131, fd35:b763:f212:0:634:78a9:db30:12cf, fe80::2004:c8f4:b08b:3c70

</details>

<details><summary>Recorder</summary>

oldest_recorder_run | 22 agosto 2025 alle ore 11:00
-- | --
current_recorder_run | 3 settembre 2025 alle ore 23:51
estimated_db_size | 1001.34 MiB
database_engine | sqlite
database_version | 3.48.0

</details>

<details><summary>Sonoff</summary>

version | 3.9.2 (a7bce04)
-- | --
cloud_online | 1 / 1
local_online | 1 / 1

</details>

### Checklist

- [x] I have enabled debug logging for my installation.
- [x] I have filled out the issue template to the best of my ability.
- [x] This issue only contains 1 issue (if you have multiple issues, open one issue for each issue).
- [x] This issue is not a duplicate issue of any [previous issues](https://github.com/ludeeus/integration_blueprint/issues?q=is%3Aissue+label%3A%22Bug%22+)..

### Describe the issue

When I updated Home Assistant the integration stopped working. The logs return an error loading Endian

### Reproduction steps

1. Update Home Assistant to 2025.9.0
2. Start Home Assistant

### Debug logs

```text
Registratore: homeassistant.setup
Fonte: setup.py:343
Prima occorrenza: 3 settembre 2025 alle ore 23:51:12 (1 occorrenza)
Ultimo accesso: 3 settembre 2025 alle ore 23:51:12

Setup failed for custom integration 'abb_powerone_pvi_sunspec': Unable to import component: cannot import name 'Endian' from 'pymodbus.constants' (/usr/local/lib/python3.13/site-packages/pymodbus/constants.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1003, in async_get_component
    comp = await self.hass.async_add_import_executor_job(
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        self._get_component, True
        ^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/local/lib/python3.13/concurrent/futures/thread.py", line 59, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1063, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/abb_powerone_pvi_sunspec/__init__.py", line 23, in <module>
    from .coordinator import ABBPowerOneFimerCoordinator
  File "/config/custom_components/abb_powerone_pvi_sunspec/coordinator.py", line 13, in <module>
    from .api import ABBPowerOneFimerAPI
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 13, in <module>
    from pymodbus.constants import Endian
ImportError: cannot import name 'Endian' from 'pymodbus.constants' (/usr/local/lib/python3.13/site-packages/pymodbus/constants.py)

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/setup.py", line 343, in _async_setup_component
    component = await integration.async_get_component()
                ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1023, in async_get_component
    self._component_future.result()
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1015, in async_get_component
    comp = self._get_component()
  File "/usr/src/homeassistant/homeassistant/loader.py", line 1063, in _get_component
    ComponentProtocol, importlib.import_module(self.pkg_path)
                       ~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/util/loop.py", line 201, in protected_loop_func
    return func(*args, **kwargs)
  File "/usr/local/lib/python3.13/importlib/__init__.py", line 88, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 1026, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/abb_powerone_pvi_sunspec/__init__.py", line 23, in <module>
    from .coordinator import ABBPowerOneFimerCoordinator
  File "/config/custom_components/abb_powerone_pvi_sunspec/coordinator.py", line 13, in <module>
    from .api import ABBPowerOneFimerAPI
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 13, in <module>
    from pymodbus.constants import Endian
ImportError: cannot import name 'Endian' from 'pymodbus.constants' (/usr/local/lib/python3.13/site-packages/pymodbus/constants.py)
```

pinogerardi, 2025-09-04:
Salve, 
non viene caricato anche su HA Container:
Home Assistant Core
2025-09-04 07:29:02.857 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration abb_powerone_pvi_sunspec which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2025-09-04 07:29:11.441 WARNING (MainThread) [homeassistant.util.loop] Detected blocking call to import_module with args ('custom_components.abb_powerone_pvi_sunspec',) in /usr/src/homeassistant/homeassistant/loader.py, line 1063: ComponentProtocol, importlib.import_module(self.pkg_path) inside the event loop; This is causing stability issues. Please create a bug report at https://github.com/home-assistant/core/issues?q=is%3Aopen+is%3Aissue
2025-09-04 07:29:11.448 ERROR (MainThread) [homeassistant.setup] Setup failed for custom integration 'abb_powerone_pvi_sunspec': Unable to import component: cannot import name 'Endian' from 'pymodbus.constants' (/usr/local/lib/python3.13/site-packages/pymodbus/constants.py)
  File "/config/custom_components/abb_powerone_pvi_sunspec/__init__.py", line 23, in <module>
  File "/config/custom_components/abb_powerone_pvi_sunspec/coordinator.py", line 13, in <module>
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 13, in <module>
  File "/config/custom_components/abb_powerone_pvi_sunspec/__init__.py", line 23, in <module>
  File "/config/custom_components/abb_powerone_pvi_sunspec/coordinator.py", line 13, in <module>
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 13, in <module>

Jump-333, 2025-09-04:
I think i solved it, from cocker you need to go to /usr/local/lib/python3.13/site-packages/pymodbus, then edit the file constants.py adding at the end this string: 
```class Endian(str, enum.Enum):
        AUTO = "@"
        BIG = ">"
        LITTLE = "<"```

Jump-333, 2025-09-04:
Solved it. You need to copy manually the files from custom_components/ (or try to upgrade the integration, but it didn’t work for me), then go to the directory /homeassistant/custom_components/abb_powerone_pvi_sunspec/. Open the file api.py, go to line 486, find slave=self._slave_id, then change it to device_id=self._slave_id.

gjlunt, 2025-09-05:
2025-09-05 17:08:58.360 ERROR (MainThread) [aiohttp.server] Error handling request from fe80::bd29:478c:43a6:76bd
Traceback (most recent call last):
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 884, in read_sunspec_modbus_model_1
    read_model_1_data = await self.read_holding_registers(
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        address=(self._base_addr + 4), count=64
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 480, in read_holding_registers
    return await self._client.read_holding_registers(
                 ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^
        address=address, count=count, slave=self._slave_id
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )  # type: ignore (pylance thinks this is not awaitable)
    ^
TypeError: ModbusClientMixin.read_holding_registers() got an unexpected keyword argument 'slave'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 714, in read_sunspec_modbus
    await self.read_sunspec_modbus_model_1()
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 913, in read_sunspec_modbus_model_1
    raise ExceptionError from exception_error
custom_components.abb_powerone_pvi_sunspec.api.ExceptionError: Unexpected error occurred

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 510, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 41, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 83, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/config/config_entries.py", line 225, in post
    return await super().post(request, flow_id)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/data_validator.py", line 74, in wrapper
    return await method(view, request, data, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/data_entry_flow.py", line 121, in post
    result = await self._flow_mgr.async_configure(flow_id, data)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 335, in async_configure
    result = await self._async_configure(flow_id, user_input)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 382, in _async_configure
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        flow, cur_step["step_id"], user_input
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 486, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/abb_powerone_pvi_sunspec/config_flow.py", line 146, in async_step_user
    uid = await self.get_unique_id(
          ^^^^^^^^^^^^^^^^^^^^^^^^^
        name, host, port, slave_id, base_addr, scan_interval
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/abb_powerone_pvi_sunspec/config_flow.py", line 119, in get_unique_id
    self.api_data = await self.api.async_get_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 688, in async_get_data
    result = await self.read_sunspec_modbus()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 782, in read_sunspec_modbus
    raise ExceptionError from exception_error
custom_components.abb_powerone_pvi_sunspec.api.ExceptionError: Unexpected error occurred
2025-

gjlunt, 2025-09-05:
Thanks, I'll check it when it wakes up tomorrow

Iminet72, 2025-09-05:
The repair solved it for me.
`address=address, count=count, device_id=self._slave_id`

Marino-ux, 2025-09-05:
@gjlunt Sorry, where can I find the beta version?

gjlunt, 2025-09-05:
Working now with version 4.0.1 beta.1

address=address, count=count, device_id=self._slave_id

marcolinux46, 2025-09-06:
I installed and modified api.py (line 481) file now with:

address=address, count=count, device_id=self._slave_id

It's working perfectly

Marino-ux, 2025-09-06:
Perfect, it works. The line could also be different, but always close to 480. Thanks.

alexdelprete, 2025-09-07:
Please test [v4.0.1-beta.3](https://github.com/alexdelprete/ha-abb-powerone-pvi-sunspec/releases/tag/v4.0.1-beta.3) and let me know if it works.

marcolinux46, 2025-09-07:
Installed v4.0.1-beta.3, i'll let you know tomorrow if it will work

Marino-ux, 2025-09-09:
no problem with restart