alexdelprete/ha-abb-powerone-pvi-sunspec #305: Unknown Error when adding Inverter

gjlunt, 2025-09-24:
### System Health details

## System Information

version | core-2025.9.4
-- | --
installation_type | Home Assistant OS
dev | false
hassio | true
docker | true
container_arch | aarch64
user | root
virtualenv | false
python_version | 3.13.7
os_name | Linux
os_version | 6.12.43-haos
arch | aarch64
timezone | Australia/Sydney
config_dir | /config

<details><summary>Home Assistant Community Store</summary>

GitHub API | ok
-- | --
GitHub Content | ok
GitHub Web | ok
HACS Data | ok
GitHub API Calls Remaining | 4994
Installed Version | 2.0.5
Stage | running
Available Repositories | 2313
Downloaded Repositories | 12

</details>

<details><summary>Solcast Solar</summary>

can_reach_server | ok
-- | --

</details>

<details><summary>Home Assistant Cloud</summary>

logged_in | true
-- | --
subscription_expiration | August 3, 2026 at 10:00 AM
relayer_connected | true
relayer_region | ap-southeast-1
remote_enabled | true
remote_connected | true
alexa_enabled | true
google_enabled | true
cloud_ice_servers_enabled | true
remote_server | ap-southeast-1-2.ui.nabu.casa
certificate_status | ready
instance_id | 4e36f669094c4e6399b8404cfa36a018
can_reach_cert_server | ok
can_reach_cloud_auth | ok
can_reach_cloud | ok

</details>

<details><summary>Home Assistant Supervisor</summary>

host_os | Home Assistant OS 16.2
-- | --
update_channel | stable
supervisor_version | supervisor-2025.09.0
agent_version | 1.7.2
[... 161 lines cut ...]

```text
Log file entry

Traceback (most recent call last):
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 821, in find_sunspec_modbus_m160_offset
    read_model_160_data = await self.read_holding_registers(
                          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        address=(self._base_addr + offset), count=1
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 489, in read_holding_registers
    raise ModbusException(
          ~~~~~~~~~~~~~~~^
        f"Device reported error: {result}",
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        address=address,
        ^^^^^^^^^^^^^^^^
        operation="read_holding_registers"
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
TypeError: ModbusException.__init__() got an unexpected keyword argument 'address'
```

alexdelprete, 2025-09-24:
That is not the debug log, but the standard HA log. I need the debug log to understand what's the issue.

gjlunt, 2025-10-01:
Issue reappeared after restart of Home Assistant

Debug log
=========================================

2025-10-01 13:12:07.009 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec] Setup config_entry for abb_powerone_pvi_sunspec
2025-10-01 13:12:07.010 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Scan Interval: scan_interval=60 update_interval=0:01:00
2025-10-01 13:12:07.024 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Coordinator Config Data: {'base_addr': 0, 'device_id': 2.0, 'host': '192.168.1.57', 'name': 'ABB Inverter', 'port': 502, 'scan_interval': 60}
2025-10-01 13:12:07.024 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Coordinator init - Host: 192.168.1.57 Port: 502 ID: 2 Base Addr.: 0 ScanInterval: 60
2025-10-01 13:12:07.025 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Data Coordinator: Update started at 2025-10-01 13:12:07.024939
2025-10-01 13:12:07.025 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (connect) [host=192.168.1.57, port=502, device_id=2, timeout=5.0]: API Client connecting
2025-10-01 13:12:07.025 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (check_port) [host=192.168.1.57, port=502]: Opening socket with 3.0s timeout
2025-10-01 13:12:07.170 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (check_port) [host=192.168.1.57, port=502]: Port connection successful
2025-10-01 13:12:07.171 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (connect): Inverter ready for Modbus TCP connection
2025-10-01 13:12:07.281 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] Connection attempt took 0.110s
2025-10-01 13:12:07.281 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (connect): Modbus TCP Client connected
2025-10-01 13:12:07.281 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (async_get_data) [device_id=2, base_addr=0]: Starting data collection
2025-10-01 13:12:07.281 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (read_sunspec_modbus_model_1) [device_id=2, base_addr=0]: Starting Model 1 read
2025-10-01 13:12:07.290 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (read_sunspec_modbus_model_1) [manufacturer=Power-One, model=-3G04-, options=5]: Device info read
2025-10-01 13:12:07.291 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (_parse_model_options) [options_string=5, opt_model_int=53]: Printable option model
2025-10-01 13:12:07.291 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (read_sunspec_modbus_model_1) [model=PVI-5000-OUTD]: Model from options
2025-10-01 13:12:07.291 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (read_sunspec_modbus_model_1) [version=C035]: Version read
2025-10-01 13:12:07.291 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (read_sunspec_modbus_model_1) [serial=794015-3G04-1015]: Serial number read
2025-10-01 13:12:07.291 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (read_sunspec_modbus_model_101_103) [device_id=2, base_addr=0]: Starting Model 101/103 read
2025-10-01 13:12:07.297 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (read_sunspec_modbus_model_101_103) [invtype_int=101, invtype_str=Single Phase]: Inverter type read
2025-10-01 13:12:07.298 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (read_sunspec_modbus_model_101_103) [total_energy=60208264, previous_energy=1]: Total energy read
2025-10-01 13:12:07.298 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (read_sunspec_modbus_model_101_103) [dc_current=-0.0, dc_voltage=-0.0]: DC values read
2025-10-01 13:12:07.298 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (read_sunspec_modbus_model_101_103) [dc_power=1683]: DC power read
2025-10-01 13:12:07.298 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (read_sunspec_modbus_model_101_103) [temp_other=36.0, temp_cabinet=0.0]: Temperature values read
2025-10-01 13:12:07.298 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (read_sunspec_modbus_model_101_103) [status=Out OC]: Device status read
2025-10-01 13:12:07.298 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (read_sunspec_modbus_model_101_103) [statusvendor=Run]: Status vendor read
2025-10-01 13:12:07.298 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (read_sunspec_modbus_model_101_103): Completed
2025-10-01 13:12:07.299 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (read_sunspec_modbus) [model=PVI-5000-OUTD]: M160 offset unknown, searching
2025-10-01 13:12:07.299 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (find_sunspec_modbus_m160_offset) [offset=122]: Searching M160 for model: PVI-5000-OUTD
2025-10-01 13:12:07.306 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (find_sunspec_modbus_m160_offset) [offset=122, multi_mppt_id=65535]: Model is not 160
2025-10-01 13:12:07.306 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (find_sunspec_modbus_m160_offset) [offset=1104]: Searching M160 for model: PVI-5000-OUTD
[... cut ...]
2025-10-01 13:12:23.784 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.api] (read_sunspec_modbus): Generic error: Unexpected error occurred
2025-10-01 13:12:23.784 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Coordinator Update Error: Unexpected error occurred at 2025-10-01 13:12:23.524695
2025-10-01 13:12:23.785 DEBUG (MainThread) [custom_components.abb_powerone_pvi_sunspec.coordinator] Finished fetching abb_powerone_pvi_sunspec (794015-3G04-1015) data in 0.259 seconds (success: False)

szmores, 2025-10-03:
My problem is similar, I access the data under node-red, with the same parameters as what I specified here.
HAOS runs under ubuntu in vmbox.

2025-10-03 14:25:20.423 ERROR (MainThread) [aiohttp.server] Error handling request from 192.168.1.11
Traceback (most recent call last):
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 896, in read_sunspec_modbus_model_1
    read_model_1_data = await self.read_holding_registers(
                        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        address=(self._base_addr + 4), count=64
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 489, in read_holding_registers
    raise ModbusException(
          ~~~~~~~~~~~~~~~^
            f"Device reported error: {result}",
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
            address=address,
            ^^^^^^^^^^^^^^^^
            operation="read_holding_registers"
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        )
        ^
TypeError: ModbusException.__init__() got an unexpected keyword argument 'address'

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 726, in read_sunspec_modbus
    await self.read_sunspec_modbus_model_1()
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 925, in read_sunspec_modbus_model_1
    raise ExceptionError from exception_error
custom_components.abb_powerone_pvi_sunspec.api.ExceptionError: Unexpected error occurred

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_protocol.py", line 510, in _handle_request
    resp = await request_handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_app.py", line 569, in _handle
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/aiohttp/web_middlewares.py", line 117, in impl
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/security_filter.py", line 92, in security_filter_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/forwarded.py", line 83, in forwarded_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/request_context.py", line 26, in request_context_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/ban.py", line 86, in ban_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/auth.py", line 242, in auth_middleware
    return await handler(request)
           ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/headers.py", line 41, in headers_middleware
    response = await handler(request)
               ^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/http.py", line 73, in handle
    result = await handler(request, **request.match_info)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/http/decorators.py", line 83, in with_admin
    return await func(self, request, *args, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
[... 13 lines cut ...]
    result = await self._async_handle_step(
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
        flow, cur_step["step_id"], user_input
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/usr/src/homeassistant/homeassistant/data_entry_flow.py", line 497, in _async_handle_step
    result: _FlowResultT = await getattr(flow, method)(user_input)
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/abb_powerone_pvi_sunspec/config_flow.py", line 146, in async_step_user
    uid = await self.get_unique_id(
          ^^^^^^^^^^^^^^^^^^^^^^^^^
        name, host, port, device_id, base_addr, scan_interval
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
    )
    ^
  File "/config/custom_components/abb_powerone_pvi_sunspec/config_flow.py", line 119, in get_unique_id
    self.api_data = await self.api.async_get_data()
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 700, in async_get_data
    result = await self.read_sunspec_modbus()
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/abb_powerone_pvi_sunspec/api.py", line 794, in read_sunspec_modbus
    raise ExceptionError from exception_error
custom_components.abb_powerone_pvi_sunspec.api.ExceptionError: Unexpected error occurred

szmores, 2025-10-04:
I don't use node-red under HA.
If I stop node-red communication your integration still gives an unknown error.

[image]

[image]

szmores, 2025-10-05:
I understand that you don't need NR for your integration. But I've been using a different system and needed it for that. I just installed HA for the first time, I haven't managed to get it working yet.
Sorry if I phrased it wrong, I'll solve the data reading differently.
Thanks for your work so far, it's useful to many people!